FreeCAD/FreeCAD #21653: Crash when deleting tree items with some LinkGroup structures

FEA-eng, 2025-05-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Forum discussion: https://forum.freecad.org/viewtopic.php?t=97355#p829276

User @dprojects noticed crashes when deleting LinkGroup -> Part -> Body -> Pad structures from the tree.

He shared a reproducible example that I attach here. Steps to trigger the crash on Linux or Access violation error on Windows:
1. Select the document in the tree
2. Press Ctrl+A to select all items in the tree
3. Pressing Delete causes crash (Linux) or Access violation error (Windows)

[FreeCAD_crash.zip](https://github.com/user-attachments/files/20516862/FreeCAD_crash.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41866 (Git)
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

Core

### Anything else?

```
OS: Ubuntu 22.04.5 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.41837 (Git) AppImage
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/100.968/1
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-31:
does it also crash in 1.0?

FEA-eng, 2025-05-31:
> does it also crash in 1.0?

I don't get any crashes or errors in 1.0.1 on Windows:

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

Syres916, 2025-06-04:
No crash using Qt5 based very recent build:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42009 (Git)
Build date: 2025/06/02 21:35:10
Build type: Release
Branch: main
Hash: 1025beed8e73d838a13bb0a3a9af3440a482a4ce
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, IfcOpenShell 0.8.2, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9245/1
Installed mods: 
  * CfdOF 1.30.4
```

Looks to be at the very least related or a duplicate of https://github.com/FreeCAD/FreeCAD/issues/20498 therefore Qt6 builds only affected.

dprojects, 2025-06-06:
> I'm building 0.21 on Qt6.8.3 for my own benefit 

If you will build the FreeCAD 0.21.2 with PySide6 please share it somewhere, because this might be the best long-term FreeCAD version ever... I was using FreeCAD 0.21.2 for a long time and I can confirm it is fast and stable version... 

In fact new FreeCAD 1.0, 1.0.1, or 1.1 is much slower and crashes a lot... in FreeCAD 1.0+ there are many problems with LinkGroup -> Part -> Body -> Pad structures... also I saw yesterday the FreeCAD 1.0.1 has broken zoom & view, sometimes the perspective view is turning on... https://www.youtube.com/watch?t=945&v=991TG7fDylI

ErtugrulEge, 2025-06-12:
I created some BIM elemnts, pressed  Shift+B  and selected all elements and pressed delete button. I get Access violation error and can't delete.

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61

[image]

closed 2025-06-16 by kadet1090