FreeCAD/FreeCAD #21655: Measurement: Selecting datum planes causes issues

FEA-eng, 2025-05-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When selecting core datum planes, the Measurement tool causes some unexpected effects:
- the plane is trimmed and its area is shown as inf km^2

[image]

- distance between the planes is not measured properly and there are errors:

[image]

```
13:01:19  <Exception> Could not get extrema
13:01:19  105.623 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 
```

The same happens with datum lines:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41866 (Git)
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Pendelf, 2025-05-30:
Yeah, something weird is going on

https://github.com/user-attachments/assets/e1e0e15c-b4c3-4713-8b07-491df735721f

FEA-eng, 2025-05-31:
> Did it work with old PD datum planes?

It's excactly the same with old datums in 1.0.1:

[image]