FreeCAD/FreeCAD #21657: Changing TC with same tool number but different spindle speed doesn't produce correct gcode

rmu75, 2025-05-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It seems the fix in #6275 (and also the test cases) got lost somewhere along the way. Tested in 1.1.0dev (see version info) and 1.0.0/1.0.1

### Full version info

```shell
OS: KDE neon 6.3 (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Austria (de_AT)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/150.702/1.39583
Installed mods: 
  * btl 0.9.9
  * kicadStepUpMod 11.3.9
  * dodo 1.0.1
  * QuickMeasure 2022.10.28
  * OpticsWorkbench 1.0.24
  * OpenTheme 2024.9.1
  * Assembly3 0.12.2
  * frame 0.1.1
  * sheetmetal 0.5.2
  * FeedsAndSpeeds 0.5.0
  * offline-documentation 1.0.0-alpha
  * InventorLoader 1.4.0
  * EM 2.1.1
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rmu75, 2025-05-30:
Testfile and generated NC code for mach3 [Tooltest.zip](https://github.com/user-attachments/files/20522234/Tooltest.zip)

rmu75, 2025-05-31:
Yes it is possible that this fix never worked.

FreeCAD used to emit tool change for every operation regardless if tool changed, so there were some redundant tool changes and spindle start/stops, but changes in spindle speed were reflected. Then this was "optimized out" in a way that looked at tool number and not tool controller. 

IMO any change in Tool Controller should emit a tool change operation, even if tool number happens to be the same. Leave any details to post processor. It is easier to remove some operation there than try to add something back.

tarman3, 2025-05-31:
I am in the dead end
In postprocessor execute this function `export(objectslist, filename, argstring)` 
But I can not find place where  `objectslist` is formed

tarman3, 2025-05-31:
@rmu75 
Please check this #21676

rmu75, 2025-05-31:
Thanks -- yes that will fix it at least for the test case. Not sure about situation with different ordering. It would be cool if you could also resurrect the automatic test cases.

Is there a reason why you don't just compare if tool controller object is a different one?

tarman3, 2025-05-31:
you mean `ToolController_1 == ToolController_2` ???

tarman3, 2025-05-31:
Maybe you are right
Need to think about it

tarman3, 2025-05-31:
But if you create few tool controllers with identical `ToolBit`, `SpindleSpeed` and `SpindleDir`
Only difference in `HorizFeed` and `VerFeed`
Probably you no need tool change commands in this case

tarman3, 2025-05-31:
Hope wil be merged #21510 and will no need to duplicate tool controller only to change `HorizFeed` and `VerFeed`

tarman3, 2025-05-31:
Of course `TC1` == `TC1` will give `True`

With this commit no matter we use one or few tool controllers
We compare properties of the tool controllers and make decision to add command change tool or not

J-Dunn, 2025-07-14:
"Is there a reason why you don't just compare if tool controller object is a different one?"

Yes, that is what is required. I frequently have multiple TC using the same tool with different feeds.

davidgilkaufman, 2025-09-05:
It's actually completely unrelated to that grant, but I wrote a PR to fix it anyway

PR #23691 (2025-09-06): [CAM] fix to ensure tool change command is issued for SpindleSpeed changes
Fixes #21657 

In summary, two tool controllers can have the same tool number but different spindle speeds. When we generate gcode, we execute a TC operation only if it is required. However, as a bug, this was implemented by checking the TC's tool number only. This means that if the user switches to a new TC with the same tool (number) but a different spindle speed, we fail to execute the TC change command.

I fix this by also checking for changes in spindle speed and direction.

As a note, if the tool (number) does not [... cut]
closed 2025-09-06 by sliptonic