FreeCAD/FreeCAD #21658: Sketcher: Sketch Corruption observed after (Sketch)MOVING[dimension update] in X(or Y)-axis was performed.

marek-stepan, 2025-05-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

HOW to reproduce/repeat issue:

Step1:  Open file  ..Stage1..... <that is file for STARTING position!>
Step2:  Press button to Edit Sketch

Possible Triggering problem with X-axis:
Step3x:  Update dimension for distance-X to 10mm only. Number is critical/important!!

Alternative Triggering problem with Y-axis:
Step3y:  Update dimension for distance-Y to  1mm only. Number is critical/important!!

Result of either Step3x/y will be sketch corruption, and RESULTS files are attached (and named ...Stage2... )

Important Beware/Warning!! Value of numbers matters! TWO comments follows:

a) Step3y:  Update dimension for distance-Y to  10mm will NOT generate any sketch corruption!!!

b) and there was provided example of observed invalid DogLeg Solver attached. Unclear why it happen during this Bug/Issue Reprot preparation.

---------------------------------------------------------

Issues were confirmed on both: 
* official FreeCad 1.0.1 release
* latest Development windows Build  1.1.0dev.41866 (Git) 2025/05/26 16:31:05

---------------------------------------------------------

Comment about ...Stage0.. file additionally provided:

File ...Stage0.. is provided to HELP changing/update STARTING position/conditions (for Sketch itself),
as this file have still 2 DoF = sketch construction is fully MOVABLE to any new needed starting position(for testing, if needed).

Difference between ..Stage0.. and ...Stage1.. is just construction and fixed dimensioning (removal of 2 DoF)

### Full version info

```shell
Confirmed on Official FreeCad 1.0.1 release:

OS: Windows 10 build 19043
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Czech/Czech Republic (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods:
  * 3D_Printing_Tools
  * AddonManager 2025.5.1a
  * FreeCAD-Documentation-html 1.0.0-alpha
  * GDT
  * kicadStepUpMod 11.7.1
  * offline-documentation 1.0.0-alpha
  * toSketch 1.0.1
  * WB_Organizer 2024.1.29

Confirmed on Latest Night-Dev-Build release:
OS: Windows 10 build 19043
Architecture: x86_64
Version: 1.1.0dev.41866 (Git)
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.6656/1
Installed mods:
  * 3D_Printing_Tools
  * AddonManager 2025.5.1a
  * FreeCAD-Documentation-html 1.0.0-alpha
  * GDT
  * kicadStepUpMod 11.7.1
  * offline-documentation 1.0.0-alpha
  * toSketch 1.0.1
  * WB_Organizer 2024.1.29
```

### Subproject(s) affected?

None

### Anything else?

This is Sketch Corrupted after just updating X-coordinate:

[image]

This is Sketch Corrupted after updating Y-coordinate:

[image]

This is Sketch OK/(not corrupted) after updating X/Y coordinates, but invalid DogLegSolver was observed:
[image]

File for INITIAL setup (may not be needed):

[IssueM02-Stage0-Object_is_Movable_2_DoF-constructionOK.zip](https://github.com/user-attachments/files/20522705/IssueM02-Stage0-Object_is_Movable_2_DoF-constructionOK.zip)

File NEEDED to repeat/reproduce this ISSUE:
[IssueM02-Stage1-FullyConstrained_0_DoF-constructionOK.zip](https://github.com/user-attachments/files/20522714/IssueM02-Stage1-FullyConstrained_0_DoF-constructionOK.zip)

Examples of ISSUE already triggered/recorded (may not be needed):

[IssueM02-Stage2-after-updating-distance-X-result-is-sketch-corruption.zip](https://github.com/user-attachments/files/20522748/IssueM02-Stage2-after-updating-distance-X-result-is-sketch-corruption.zip)

[IssueM02-Stage2-after-updating-distance-Y-result-is-sketch-corruption.zip](https://github.com/user-attachments/files/20522758/IssueM02-Stage2-after-updating-distance-Y-result-is-sketch-corruption.zip)

[IssueM02-Stage2-after-updating-distance-Y-result-is-OK-just-Invalid_DogLegSolver.zip](https://github.com/user-attachments/files/20522772/IssueM02-Stage2-after-updating-distance-Y-result-is-OK-just-Invalid_DogLegSolver.zip)

I believe that BEFORE much more complex coordinates/dimension TRANSLATE issues like THIS one: https://github.com/FreeCAD/FreeCAD/issues/21553  ) will be solved, I wish my issue to be fixed FIRST, because it is EASIER IDENTIFIED and SEPARATED Issue/impact setup --> This Issue is much easier to repeat, reproduce and therefore it is easier to understand (DEBUG and ANALYZE) what is going wrong..... 

Thank you very much in advance.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-05-30:
Hi @marek-stepan, I am trying to reproduce the issue. I opened the sketch in edit mode. 

> Step3x: Update dimension for distance-X to 10mm only. Number is critical/important!!

Now which parameter do you mean when you say "distance-X" in the following screenshot?

[image]

marek-stepan, 2025-05-30:
[image]

Katt-Kushol, 2025-05-31:
The issue is repeatable. An extreme change in the coordinate appears to modify random arcs in the above sketch. It appears to be in the same category of "sketch flipping" issue.

[image]

Version Details:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.9548/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * sheetmetal 0.7.22
  * Silk 0.1.7

maxwxyz, 2025-05-31:
@AjinkyaDahale FYI

AjinkyaDahale, 2025-12-09:
Still repeatable. There are other similar issues and unfortunately I still don't have a good idea for how this can be resolved reliably.

```
OS: Ubuntu Core 24 (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.2.0dev.44586 (Git) Snap 1966
Build date: 2025/12/07 10:43:09
Build type: Release
Branch: main
Hash: 915b6f990167c5840346e6de6be99d88dec40ad0
Python 3.12.3, Qt 6.9.1, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 6.9.3
shiboken 6.9.3, SMESH 7.7.1.0, xerces-c 3.2.4, IfcOpenShell 0.8.4, OCC 7.8.0
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/147.909/1.09375
```