FreeCAD/FreeCAD #21660: Hole size is incorrect when i create a hole on a surface of my part

RandyHut, 2025-05-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I created a disc.. Then selected the top surface, sketched two circles @.125 Dia(not rad.), dimension-ed them and closed my sketch. then using the Hole tool, it creates .250Dia holes. this is with the weekly build 41866 conda-linux-x86_64-py/311 appimage. running on MX Linx 23.6 that is up to date.  

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41866 (Git) AppImage
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/69.9239/1
```

### Subproject(s) affected?

None

### Anything else?

I searched for this and could no find any info on it. Hope it is ok i posted it

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-05-30:
The Hole tool does not care about the diameter/radius of the circles. It just uses their centers.
Please read:
https://wiki.freecad.org/PartDesign_Hole#Description

Closing this one.

closed 2025-05-30 by Roy-043

NonaSuomy, 2025-08-10:
@RandyHut  https://forum.freecad.org/viewtopic.php?t=98989