FreeCAD/FreeCAD #21662: Sketcher>UI - The toolbars don't remember their locations when docked in the StatusBar.

FreeCAD-Nut, 2025-05-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The toolbars don't remember their locations when docked in the StatusBar. Upon restart, their location will be reset in the top bar. The problem seems to affect only the Sketcher workbench.

1. In an active Sketcher session drag a toolbar to the StatusBar (see the attached screenshot).
2. Restart FreeCAD.
3. Notice the toolbar is missing in the StatusBar.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FreeCAD-Nut, 2025-05-31:
You are right. I tested extensively all toolbars available in the Sketcher WB and noticed the problem affects only the Sketcher native toolbars. Additionally, I found the Sketcher native toolbars cannot be undocked. Upon restart, FC will dock them automatically to the Top bar. This is the list of the problematic native toolbars:

Sketcher edit tools
Sketcher visual
Sketcher B-spline tools
Sketcher tools
Sketcher constraints
Sketcher geometries
Sketcher edit mode

[image]

luzpaz, 2025-09-26:
@FreeCAD-Nut is this still the case in latest 1.1-dev ?