FreeCAD/FreeCAD #21667: BIM: Site terrain does not toggle visibility when site is toggled

Harvie, 2025-05-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I create BIM site and attached terrain model to it. It starts behaving in a weird way. If i toggle site visibility, the terrain is unaffected, which is super annoying, because i need to toggle it manualy each time i want to toggle site.

Also when i set the terrain to the site it is moved into the site subtree, but when it's already in the subtree before i add it, it's there twice after addition and occasionaly gives duplicate child errors.

[site-toggle.FCStd.zip](https://github.com/user-attachments/files/20528434/site-toggle.FCStd.zip)

### Full version info

```shell
OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.1.0dev.41837 (Git) AppImage
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Turntable/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 90/172.811/1
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-31:
Do you mean the rectangle? @Roy-043 is this a Draft or BIM issue?

Harvie, 2025-05-31:
This is BIM issue. I mean the rectangle, but it happened even when terrain was any other kind of object (eg. loft between sketches).
There is some unusual way that BIM uses to handle terrain model of a site, perhaps it needs some tuning.

semhustej, 2025-05-31:
I am not sure this is a bug. 2D objects, which are children (mostly base objects) of BIM objects, hide and unhide independently of the parent object. You can see that when trying to hide/unhide the wall and it's base object - the sketch. It behaves the same way as the site - rectangle pair.

See behavior in the video:

https://github.com/user-attachments/assets/3f360ffe-dba3-4660-b15e-4737de4f8476

As a workaround you can create a slab or an extrusion from the rectangle and than it hides/unhides together with the site.

It would create a lot of unwanted geometry being visible, if base objects were unhidden together with the parent BIM objects. The question is, whether Site object should behave the same way.

Harvie, 2025-05-31:
> I am not sure this is a bug. 2D objects, which are children (mostly base objects) of BIM objects

When i create loft (is that 3d object by your terms? it's not in single plane, so it's 3D in mine terms) it behaves exactly the same. i've used rectangle here just to provide minimal viable example.

Also i base all my walls and roofs on sketches and they do toggle along with 3D models (eg. when i toggle site, building or floor). Which is super annoying, had to evict all of the base sketches to construction group... Terrain on the other hand is part of 3D model that i usualy want to show.

semhustej, 2025-05-31:
> When i create loft (is that 3d object by your terms? does not have thickness, but it's not in single plane, so it's 3D in mine terms) it behaves exactly the same. i've used rectangle here just to provide minimal viable example.

Maybe the distinction is solid vs non-solid objects.

Harvie, 2025-05-31:
> Maybe the distinction is solid vs non-solid objects.

Nope. There is not distinction. Does exactly the same thing for cube from part workbench and even for my loft converted to solid...

Roy-043, 2025-06-01:
The Rectangle and the Wall have a different relation with the Site. The Wall is in the Group. The Rectangle is referenced by the Terrain property of the Site. In the representation in the Tree view this distinction is lost.

Roy-043, 2025-06-01:
The Terrain should always stay invisible. The Site derives its own shape from the Terrain (and possible additions and subtractions).

Harvie, 2025-06-01:
> In most cases the Terrain should always stay invisible. The Site derives its own shape from the Terrain.

Oh sorry for misleading you. i am talking about this "site shape" this whole time...

Roy-043, 2025-06-01:
The shape of the Site in the 3D view should become (in)visible with the Site object. Make sure the Terrain object is (and stays) invisible and try changing the visibility of the Site.

Harvie, 2025-06-01:
It behaves almost erraticaly. Now i had the the terrain within the site tree, but when i assigned it as terrain to the site, it's there twice and it toggles with the site, but keeps giving me this:

```
22:40:44  10.0645 Tree.cpp(295): [ComboView] duplicate child item krum_v7#Site.Loft
22:42:21  107.585 Tree.cpp(295): [ComboView] duplicate child item krum_v7#Site.Loft
22:42:22  108.113 Tree.cpp(295): [ComboView] duplicate child item krum_v7#Site.Loft
```
On the other hand when it was not there, it was copied/linked to the site when i assigned it as terrain, but it did not rendered correctly until i recomputed the site, which caused more confusion (because that made me think i need to have that object visible in order to see the terrain).

It's just super random and confusing, i can't even list all the possible combinations of tiny workflow nuances that cause different outcomes when using this thing.

Harvie, 2025-06-10:
Now it seems to behave consistently regardless of base terrain geometry being in the site group or not...

```
OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Turntable/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 90/172.811/1
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58

```