FreeCAD/FreeCAD #21669: Draft_Dimension does not work with Building US unit system

semhustej, 2025-05-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open new file
- change unit system to `Building US`
- start `Draft_Dimension`
- try to create the dimension. It is not created and there is an error message

<details><summary>Error message</summary>

```
15:08:54  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_dimension.py", line 365, in attach
    self.updateData(vobj.Object, "Start")
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_dimension.py", line 611, in updateData
    self.string = units.display_external(length,
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftutils\units.py", line 131, in display_external
    display = fmt.format(float(internal_value) / float(conversion))
                         ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
<class 'ZeroDivisionError'>: float division by zero
15:08:54  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_dimension.py", line 725, in onChanged
    self.updateData(obj, "Start")
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_dimension.py", line 611, in updateData
    self.string = units.display_external(length,
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftutils\units.py", line 131, in display_external
    display = fmt.format(float(internal_value) / float(conversion))
                         ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
<class 'ZeroDivisionError'>: float division by zero
15:08:54  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_dimension.py", line 725, in onChanged
    self.updateData(obj, "Start")
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_dimension.py", line 611, in updateData
    self.string = units.display_external(length,
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftutils\units.py", line 131, in display_external
    display = fmt.format(float(internal_value) / float(conversion))
                         ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
<class 'ZeroDivisionError'>: float division by zero
15:08:54  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_dimension.py", line 725, in onChanged
    self.updateData(obj, "Start")
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_dimension.py", line 611, in updateData
    self.string = units.display_external(length,
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftutils\units.py", line 131, in display_external
    display = fmt.format(float(internal_value) / float(conversion))
                         ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
<class 'ZeroDivisionError'>: float division by zero
15:08:54  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_dimension.py", line 611, in updateData
    self.string = units.display_external(length,
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftutils\units.py", line 131, in display_external
    display = fmt.format(float(internal_value) / float(conversion))
                         ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
<class 'ZeroDivisionError'>: float division by zero
15:08:54  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftviewproviders\view_dimension.py", line 611, in updateData
    self.string = units.display_external(length,
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-41866-conda-Windows-x86_64-py311\Mod\Draft\draftutils\units.py", line 131, in display_external
    display = fmt.format(float(internal_value) / float(conversion))
[... 107 lines cut ...]
Architecture: x86_64
Version: 1.1.0dev.41866 (Git)
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
```

### Subproject(s) affected?

Draft

### Anything else?

This regression has been added recently. I don' think it's older than 14 days.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-06-03:
@semhustej I'm testing a fix for this as part of https://github.com/FreeCAD/FreeCAD/pull/21147 the fact the author was able to reproduce gives confidence in the fix

@maxwxyz could _Type: Regression_ label be added to this please to assist searching, TIA

semhustej, 2025-06-08:
Tested with latest weekly and the issue is fixed. Thank you.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1

```