FreeCAD/FreeCAD #21671: BIM: Arch_Pipe not created correctly

semhustej, 2025-05-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open new file
- create `Draft_Line`
- select the line and start the `Arch_Pipe` command
- an object is created but is not correct Arch_Pipe

See behavior in the video:

https://github.com/user-attachments/assets/19565df7-b16f-4aab-b302-2bf2835346b5

I am not able to reproduce with FC 1.0.1

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41866 (Git)
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
```

### Subproject(s) affected?

BIM

### Anything else?

I think the regression was introduced in the last 14 days after I posted this message.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-05-31:
Confirmed in 
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41905 (Git)
Build date: 2025/05/30 18:02:58
Build type: Release
Branch: main
Hash: 239a05b242b9162fd7a5ab13dcfa86b097c5ad55
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods: 
  * Cables 0.1.4
  * CfdOF 1.30.1
  * CurvedShapes 1.0.13
  * Curves 0.6.61
  * dodo 1.0.1
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Gridfinity 0.11.7
  * Manipulator 1.5.9
  * OpticsWorkbench 1.2.0
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.22
  * Silk 0.1.5
  * Telemetry 1.0.0
```

tetektoza, 2025-05-31:
Will leave it for @furgo16 as it seems his changes will probably fix this one too.