FreeCAD/FreeCAD #21689: GUI 3D view not working correclty after dialog window close

zisco, 2025-06-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

After the closing of an application dialog window, the area of the 3D view corresponding to the area covered by the dialog window is inaccessible to the user, as if the dialog were "still there": left button click is not responsive and drag stops at the area boundary. The issue is reproducible also in safe mode. 

In the uploaded video, it is first shown random draft objects correctly created, then a dialog window opened and closed, and finally a second round of draft object creation. You can see that in the second round no object can be created in the area previously covered by the dialog.

https://github.com/user-attachments/assets/1d26e261-345f-4899-b281-54fd600c41c4

### Full version info

```shell
OS: Fedora Linux 42 (Workstation Edition) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.41905 (Git) AppImage
Build date: 2025/05/30 18:02:58
Build type: Release
Branch: main
Hash: 239a05b242b9162fd7a5ab13dcfa86b097c5ad55
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/161.962/1
Installed mods: 
  * dodo 1.0.1
  * OpticsWorkbench 1.2.1
  * Reinforcement
  * Curves 0.6.62
  * FreeCAD_SketchArch
  * freecad.optics_design_workbench 0.6.2
  * sheetmetal 0.7.22
  * ExplodedAssembly
  * fasteners 0.5.38
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

OV3RDOSE, 2025-06-01:
Same in flatpak on fedora 42...

OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 878f0b8)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

OV3RDOSE, 2025-06-02:
Maybe a problem with Mutter...
After updating, I no longer have any problems.
See here:
[https://gitlab.gnome.org/GNOME/mutter/-/issues/4133](https://gitlab.gnome.org/GNOME/mutter/-/issues/4133)

zisco, 2025-06-02:
> Maybe a problem with Mutter... After updating, I no longer have any problems. See here: https://gitlab.gnome.org/GNOME/mutter/-/issues/4133

Apparently you are right. Tried to downgrade mutter (48.3 --> 48.1) and I have no longer the problem.

OV3RDOSE, 2025-12-15:
No, that's no longer an issue. Mutter has been fixed.