FreeCAD/FreeCAD #21690: Spreadsheet column wide is fixed

yahbluez, 2025-06-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In this version the wide of columns can not be changed and is fixed to a very tiny default size.

[OT]
It would be very user friendly if **any** pop up window would save the last size (and position) the user selected for it.

### Full version info

```shell
OS: KDE Flatpak runtime (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40267 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 04d2bcb)
Hash: 04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical/physical DPI: 96/114.694
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-06-01:
Can you test again with a more recent weekly? Can't confirm with:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41905 (Git)
Build date: 2025/05/30 18:02:58
Build type: Release
Branch: main
Hash: 239a05b242b9162fd7a5ab13dcfa86b097c5ad55
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

yahbluez, 2025-06-01:
@Roy-043 tested this version:
OS: Manjaro Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41905 (Git) AppImage
Build date: 2025/05/30 18:02:58
Build type: Release
Branch: main
Hash: 239a05b242b9162fd7a5ab13dcfa86b097c5ad55
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Turntable/Object center
Stylesheet/Theme/QtStyle: unset/Dark theme/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/114.694/1

Now the columns can be changed in size but the window size modifiers do not work at all.

If i had a wishlist - ever since i use freecad - I would like to have a cleaner window handling that just saves the settings for each window instead of *toggle between bad or worse* switches. 

It is so easy in QT to store and save the window settings a user made. Everytime i save or load something in freecad i have to adjust the file selector window again because it is so tiny that selection is hard to do.

luzpaz, 2025-06-01:
> If i had a wishlist - ever since i use freecad - I would like to have a cleaner window handling that just saves the settings for each window instead of toggle between bad or worse switches.
>
> It is [...]

Worth opening a separate ticket for this.

yahbluez, 2025-06-02:
@xtemp09 if you hover the mouse at the window border you can grap the border and resize the window.
That broken somehow in one of the dev versions i tried to use.
I guess side effects of the very very unusual way freecad handles windows.

yahbluez, 2025-06-02:
@xtemp09 no i was not able to grap the windows inside freecad also resize did not work. But this was only happen in one dev version the newer version i use now didn't have that issue. In my opinion the whole window handling needs a rethinking. Toggel between two states is a very worse idea just take the settings the user made and everything is good.

xtemp09, 2025-06-02:
@yahbluez, are you saying that everything works as expected in the latest weekly bundle and a release build? Perhaps, the issue should be modified or a separate issue should be created, concerning window toggling.

yahbluez, 2025-06-03:
Issue fixed since dev version:

OS: Manjaro Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/Dark theme/Fusion

yahbluez, 2025-06-03:
.

maxwxyz, 2025-06-03:
Thanks, closing then.

closed 2025-06-03 by maxwxyz