FreeCAD/FreeCAD #21699: Simple Pipes of Linear Segments Fail

dougransom, 2025-06-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The pipe may not complete for the last segment in a some vertical segments.   Breaks completely saving and reopening a file with the problem.

In report view, this is displayed on opening the file
 `09:39:55  <Part> ViewProviderExt.cpp(1308): Cannot compute Inventor representation for the shape of netcuptrellis#Body: Bnd_Box is void
09:39:55  <Part> ViewProviderExt.cpp(1308): Cannot compute Inventor representation for the shape of netcuptrellis#AdditivePipe: Bnd_Box is void
09:39:55  <Part> ViewProviderExt.cpp(1308): Cannot compute Inventor representation for the shape of netcuptrellis#Body: Bnd_Box is void
09:39:55  4260.39 <App> Document.cpp(2163): Failed to restore netcuptrellis#Sketch005.ExpressionEngine: Failed to parse expression.
` 

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.38
  * ThreadProfile 1.96.0
```

### Subproject(s) affected?

None

### Anything else?

The simple freecad drawing shows the problem.  Check out the reportview errors after loading the file.  The Pipe doesn't seem to work.

[netcuptrellis.zip](https://github.com/user-attachments/files/20541354/netcuptrellis.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-06-01:
Change the "Transition" property of the Pipe to "Right corner".

dougransom, 2025-06-02:
The problem has something to do with the last point on the path, which in my example, was a reflection of the first point on the path around the vertical axis.   I worked around the problem by moving the last point in the path up about 1/2mm (so it ended in the middle of the  base feature instead of on the face).   

I do believe this is a bug both in the error messages not being to helpful and that the pipe doesn't work.

Roy-043, 2025-07-07:
With the "Transition" property set to "Transformed" there is a self-intersection at the apex.The error messages are indeed unclear.