FreeCAD/FreeCAD #21701: No effect of X and Y attachment offset on sketch attached to tilted cylinder

theosib, 2025-06-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. Go to Part workbench and make a cylinder.
2. Rotate the cylinder around the X axis.
3. Attach a sketch to the top face of the cylinder.
4. Import the external geometry of the face and exit Sketcher.
5. Go to the attachment offset of the sketch and try adjusting Position.

Expected:
- All position axes should affect the position of the sketch relative to the attachment.

Actual:
- Only the Z offset has any effect. The X and Y are ignored.

### Full version info

```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.41837 (Git)
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/Classic/macos
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * lattice2 1.0.0
  * Alternate_OpenSCAD 1.0.0
  * Curves 0.6.50
```

### Subproject(s) affected?

Core

### Anything else?

See attachment for example.
[cylsketchoffset.FCStd.zip](https://github.com/user-attachments/files/20541931/cylsketchoffset.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

theosib, 2025-06-02:
There is definitely a misunderstanding.

I totally expect that the the position offset is relative to the sketch plane. This isn't happening.

When setting the sketch position, it's NOT MOVING AT ALL IN ANY DIRECTION.

Please reopen this bug report since I don't my observation was understood.

Thanks.

theosib, 2025-06-02:
In the first image, you can see the circle on top of the cylinder with no offset.
[image]

If I change the Z offset, it moves away from the supporting surface as expected.
[image]

But in this next image, I've additionally changed the X offset to 10 meters. Notice that it hasn't moved.
[image]

This is a new bug.

I have many times attached sketches to surfaces and then set an offset relative to that attachment. It hasn't been until this snapshot that I've noticed that X and Y offsets no longer work.

theosib, 2025-06-02:
To prove my point, I went back to 1.0.0 and did the same thing, and the offset works just fine.
[image]

Please see the attachment that works fine in 1.0.0.
[cyltest2.FCStd.zip](https://github.com/user-attachments/files/20554207/cyltest2.FCStd.zip)

I can create this in 1.0.0 and load it into the latest snapshot, and it still works. But I can't *create* it in the latest snapshot.

theosib, 2025-06-02:
I've continued playing with this, and I think I've figured out what's gone wrong. And yes, this is definitely a regression.

Consider three kinds of geometry in a sketch:
1. Imported geometry that is not construction.
2. New geometry that is synchronized to imported construction geometry.
3. New geometry that is only positioned relative to the sketch's origin.

Just to be clear, here's how to make each one:
1. Ensure construction geometry is not enabled, select the import tool, and select an object.
2. Ensure construction geometry IS enabled, select the import tool, and select an object. Then create another piece of geometry and set constraints on the new one relative to the import.
3. Create new geometry and constrain it relative to the origin.

When creating geometry of either type 2 or 3, changing the sketch's attachment offset works *exactly as expected.*

However, if you create type 1 geometry, attachment offset in X or Y *doesn't work at all.*

Just to see how bizarre this is, create a sketch with both type 1 and type 3 geometry. Next change the attachment X or Y offset using the little spinner. Notice how the whole sketch moves in sync as expected. But when you press Tab or otherwise deselect that axis, the type 1 geometry and ONLY the type 1 geometry snaps back into its old position.
[image]

The bug here is violation of the perfectly reasonable expectation that when you change the attachment offset of a sketch, the WHOLE THING should move in sync, not just parts of it.

Roy-043, 2025-06-02:
> Please see the attachment that works fine in 1.0.0.

But the circle that you see moving is NOT external geometry. You have added a normal circle with a center constrained to the origin of the sketch. Please open the sketch after changing the X and Y offsets and notice that the external geometry circle no longer coincides with the circle you have added.

theosib, 2025-06-02:
I really appreciate your patience with me on this. It's very surprising to me to see sketch elements not move with their attachment offset. But if this is intentional, I guess I just have to get used to it.

Previously, I thought I was constraining the second circle to the center of the imported circle, but instead I had accidentally constrained to the origin of the sketch. Since those are coincident, and it's hard to tell them apart.

Using the Elements panel, I managed to select the center of the imported circle and then constrained another circle to that. THIS one snaps back to the attachment origin.