FreeCAD/FreeCAD #21725: CAM: Engrave strange down movements

tarman3, 2025-06-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In all depths before last get strange movement down in the end of path

https://github.com/user-attachments/assets/038fbb20-03f1-48fb-84cf-0da2ce0a99fa

This easy to fix, but this is look not like a mistake, since part of the code is added to add this strange movements

https://github.com/FreeCAD/FreeCAD/blob/b0e901bab6c334bcdb5f0037c0d9f1f95cad397c/src/Mod/CAM/Path/Op/EngraveBase.py#L86-L92

Can someone explain why we need this extra movements?

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/118.293/1
Installed mods: SearchBar 1.4.2 (Disabled); CurvedShapes 1.0.13; lattice2 1.0.0; Silk 0.1.9; FeedsAndSpeeds 0.6.0 (Disabled); freecad.gears 1.3.0; btl 0.9.9 (Disabled); fasteners 0.5.38; MeshRemodel 1.10.37; Curves 0.6.62; sheetmetal 0.7.22; SaveAndRestore 0.2.1 (Disabled); Defeaturing 1.2.6
```

### Subproject(s) affected?

None

### Anything else?

https://forum.freecad.org/viewtopic.php?t=97468

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-06-02:
@shaise when you get a chance, do you mind weighing-in ?

tarman3, 2025-06-02:
@sliptonic , maybe you know the answer?

tarman3, 2025-06-02:
> Looks like a mistake/regression to me.

I thought the same, after I saw gcode

But after opened `Mod/CAM/Path/Op/EngraveBase.py` changed my opinion
To much code to be a mistake

Maybe this movement must be in opposite direction (up)?

luzpaz, 2025-06-03:
@shaise my bad...thought it was a simulator issue initially

PR #21732 (2025-06-24): CAM: Engrave - Fix step down for closed profile
Fix #21725
https://forum.freecad.org/viewtopic.php?t=97468

Did not remove the code
Maybe someone will be able to clarify in which cases this code should work