FreeCAD/FreeCAD #21736: FEM:  Direction setting for the constraining force over the surface is broken

rhabacker, 2025-06-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In this version, it is not possible to set the direction of a force constraint via the surface

## How to reproduce
1. execute the following content of a recorded macro to add a new document with an associated test case and open the constraint setup dialog:
```
import FreeCAD
import importlib
import femexamples.truss_3d_cs_circle_ele_seg3 as example
importlib.reload(example)
example.setup()
App.setActiveDocument("Unnamed")
App.ActiveDocument=App.getDocument("Unnamed")
Gui.ActiveDocument=Gui.getDocument("Unnamed")
App.getDocument('Unnamed').getObject('LoadLine').ViewObject.Visibility = True
App.getDocument('Unnamed').getObject('ConstraintForce').ViewObject.doubleClicked()
```
As the recorder has not recorded any keystroke events, the following steps must be carried out manually.

2. Press `tab` 3 times (to enter the input field for the direction)
3. clear the input field
4. select the “LoadLine” object in the 3D view or in the model window

The name of the selected edge `LoadLine:Edge1` is not inserted in the input field, it must be entered manually.

### Full version info

```shell
OS: openSUSE Leap 15.6 (KDE/plasma5/xcb)
Architecture: x86_64
Version: 1.1.0dev.42006 (Git) AppImage
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/130.665/1.25
Installed mods: 
  * Assembly4 0.50.18
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rhabacker, 2025-06-03:
Just a few additional information:

1. objects are added to the list to which the restriction is to be applied by pressing the “Add” button and then selecting objects. When the button is pressed, it is framed in color, presumably to indicate that this function is active.
2. if you click on the input field for the direction, this is also framed in color, but you cannot add an object to it, which in my opinion is inconsistent and therefore confusing.

FEA-eng, 2025-06-03:
> 2. Press `tab` 3 times (to enter the input field for the direction)
> 3. clear the input field
> 4. select the “LoadLine” object in the 3D view or in the model window
> 
> The name of the selected edge `LoadLine:Edge1` is not inserted in the input field, it must be entered manually.

After selecting the edge, you have to press the Direction button. Then the name of the edge appears in the input field:

[image]

rhabacker, 2025-06-05:
This already exists within the FEM module for example in the TaskFem* dialogs, see the attached screenshot from designer.
```
 cd ~/src/FreeCAD
 find src/Mod/Fem/Gui/ -name '*.ui' | grep TaskFem | xargs designer
```

[image]

luzpaz, 2025-06-05:
Just an aside, that's ingenius way to show all specified `.ui` files in QtDesigner, bravo!

luzpaz, 2025-08-14:
What's the verdict here ?

rhabacker, 2025-08-15:
> One important detail I noticed after you posted the animation is that FreeCAD was used on a KDE desktop with a dark theme, where the input field looks exactly like a button, as shown in the screenshot below. [image]

The real problem is that after opening the dialog box and selecting the direction button with the Tab key, the focus is on the input field and not on the button. I updated the subkect accordingly

luzpaz, 2025-08-16:
@FEA-eng do you concur ?

FEA-eng, 2026-08-27:
No changes in this regard