FreeCAD/FreeCAD #21755: BIM: Array with levels and walls does not take into account nesting.

kaiwas, 2025-06-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am looking for some way to create many typical floors of buildings in a parametric way.

The most potentially useful tool is creating arrays of levels. When creating an array, the levels are arranged in the desired way, but do not include walls and windows. The extended array setting does not affect the result.
In addition, in an extended array, the elevation of each level does not change.

At the same time, if I multiply the walls without levels by an array, I get an array of walls, but do not see windows and doors.

It seems that the array for "container objects" does not take into account nesting in these containers.

Demonstration on video.

https://github.com/user-attachments/assets/6d8f3d7f-8632-4fb8-b3eb-a1b1f1eb0a51

Maybe there is some other way to quickly multiply identical floors so that they have connections between themselves, but I have not found such a way. I will be glad to any tips.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42006 (Git) AppImage
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.63
  * freecad.gears 1.3.0
  * TrackingGroups.zip
  * sheetmetal 0.7.22
  * OpenTheme 2025.5.20
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-06-04:
> Clone Level Up does nothing for me at all. Besides, it's a one-time operation (probably). 

If the command is not working for you, we might want to discuss it at https://forum.freecad.org/viewtopic.php?p=829125#p829125

> But what if there are 10 identical floors and at some point you need to increase or decrease their number? An array would be best suited for this.

I can see the use case, I'm just trying to suggest some workarounds. Perhaps adding a new command is the answer, but that comes with more complexity and more code to maintain. Hence the suggestion to exhaust all other possibilities first.

> Perhaps these words may seem harsh, but it is true - in FreeCAD everything falls apart or does not do what it should do. But for some reason it is important to everyone that the IFC structure is supported. Why? To create models in another program and view them in FreeCAD? This is sad.
It would be great if IFC was equal to FreeCAD, and not vice versa )))

I can only speak for me personally:

1. There are parts of the BIM code that are well structured, but others have grown organically. It's just natural in open source projects. As someone who is a recent contributor, the complexity of the later make it exceedingly difficult to fix issues or add new features. First, you have to understand the (20-year-old) code, then you need to understand all cases and edge cases that are being supported. Then you need to support those cases and possibly intermingle them with new ones you are adding in your changes. Finally, you need to make sure you are not introducing regressions. As a humble effort to improve the situation, I started to submit refactoring contributions to simplify the code in some areas. But mostly, to introduce unit tests to catch regressions before they are merged into `main` (https://github.com/FreeCAD/FreeCAD/pull/21342, https://github.com/FreeCAD/FreeCAD/pull/21541). Whether IFC native is a good addition or not should probably be discussed elsewhere (I'm neutral to it, as I only use it for export/import). But the fact is that the feature is there, and it has to be taken into consideration. If the use case in this issue is a common one or a very specialized one should also be weighed in.
2. My use case for IFC is interoperability and supporting open standards. That's also one of the main reasons I started to contribute to open source many years ago. For the more practical reason, it allows me to work with (you could name them old-school, not in a derogatory way) architects who mostly work with 2D. They might not have a full 3D workflow, but IFC allows me to share models with them in an (open) format they can open and examine with their proprietary tools.

> in FreeCAD everything falls apart or does not do what it should do

But that's not because of IFC! :-) In any case, let's keep working together to improve and build great things.

kaiwas, 2025-06-04:
> But that's not because of IFC! :-) In any case, let's keep working together to improve and build great things.

I do not reject the IFC. It is just that in my opinion it should be a secondary issue. It is certainly good if it is possible to take into account the work with this format, but it should not be a limitation in development.

furgo16, 2025-06-04:
I think we agree, then :-)