FreeCAD/FreeCAD #21768: Sketcher fails with "illegal character in mapped name" message

carribeiro, 2025-06-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Creating new sketches with the latest two weekly builds of FreeCad fails. Sometimes I get it to work but it's not reliable. The following error message appears:

11:34:19  <Exception> ElementMap.cpp(570): Illegal character in mapped name: ష�
1;SKT
11:34:19  Sketch: Illegal character in mapped name: ష�
1;SKT
11:34:19  <Exception> ElementMap.cpp(570): Illegal character in mapped name: ష�
1;SKT
11:34:19  Sketch: Illegal character in mapped name: ష�
1;SKT

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/06/02 18:42:58
Build type: Release
Branch: main
Hash: 997d485124fab7b9ba971df59e91b55aeb5de61a
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Portuguese/Brazil (pt_BR)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/106.57/1.5
Installed mods: 
  * Curves 0.6.62
  * fasteners 0.5.38
  * FreeGrid 2.2.0
  * LCInterlocking
  * Manipulator 1.6.0
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

carribeiro, 2025-06-04:
I've tried to edit the same file I created in the previous example with 1.0.1, and editing the sketch created before failed (with a different message). However creating a new sketch worked fine.

The error when trying the same sketch created on the newer weekly release on the 1.01. stable was this:

12:09:01  Sketch001: AttachEngine3D: attached to a non App::GeoFeature 'Origin'
12:09:09  Sketch001: AttachEngine3D: attached to a non App::GeoFeature 'Origin'
12:09:12  Sketch001: AttachEngine3D: attached to a non App::GeoFeature 'Origin'

carribeiro, 2025-06-04:
Steps to reproduce:

1. Start FreeCAD, create a new file, new body, create sketch on XY Plane.
2. Draw ANYTHING, even a simple line is sufficient.
3. The error message appears on the report view.
4. Sometimes the sketch disappear or is shown up partially.

My best guess is that this is something locale-dependent.

Katt-Kushol, 2025-06-05:
Hi, I have seen similar warning/error notifications about a month ago. See here: [https://forum.freecad.org/viewtopic.php?p=825673#p825673](url) 
Later, I figured it was the faulty weekly built for Windows which came out that time. The windows build was having issue for several weeks in May 2025. This makes me wonder, if the error you are having has the same origin: something with the weekly built.

carribeiro, 2025-06-05:
I just recently moved from the older dev releases located at https://github.com/FreeCAD/FreeCAD-Bundle/releases to the new location at https://github.com/FreeCAD/FreeCAD/releases. That may be the case indeed.

maxwxyz, 2025-06-05:
Windows builds on the main repo are still having issues. If there is an issue in the one from https://github.com/FreeCAD/FreeCAD-Bundle/releases  please comment and I'll reopen.
@oursland FYI

closed 2025-06-05 by maxwxyz