FreeCAD/FreeCAD #21772: FC 1.0 crashes when escape is pressed on MacOS Sequoia

tankshield, 2025-06-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Freecad crashes very often when the escape button is pressed. For example, when a tool is selected and then after using it, "esc" button is pressed, FC 1.0 Crashes. It has crashed so many times that it seems like it is non-production ready. 

Intel chipset on MacOS Sequoia. 

Crash Report is attached.

### Full version info

```shell
OS: macOS 15.5
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * EasyProfileFrame 0.0.1
  * addFC 1.5.2
  * Render 2024.12.15
  * boltsfc 2022.11.5
  * ArchTextures
  * Quetzal 1.5.6
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-06-05:
@johnocfii any chance you can test?

tankshield, 2025-06-07:
Hi, @luzpaz for pointing out that there is an updated version. I tested it out with 1.0.1 and was not able to reproduce it. But haven’t done extensive work on it so far. I’ll open another issue if it surfaces again. Closing this as solved in version 1.0.1 for now. Thanks all for the help.