FreeCAD/FreeCAD #21782: [UI/UX] Can't see half of the expression editor when properties tab alignet to right

ErtugrulEge, 2025-06-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings everyone!
I alignet properties panel to right and when I press expression editor button I can't see half of it

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-06-05:
I can confirm the issue.

It is possible to move the expression editor, so it is not limiting the functionality. But the default location is problematic.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

semhustej, 2025-06-06:
I can reproduce the issue with single monitor setup.

ErtugrulEge, 2025-06-06:
> [@ErtugrulEge](https://github.com/ErtugrulEge) Are you using two monitors?

No, I have only one monitor

B0cho, 2025-06-06:
What should be expected behaviour?
Would it be, that this Dialog is opened a bit closer into screen center, no matter which panel it is located?

Pendelf, 2025-06-06:
@B0cho 

I'm duplicating the video from https://github.com/FreeCAD/FreeCAD/issues/18119 

https://github.com/user-attachments/assets/cc11ffa1-7865-4cdf-bc7c-1622dfaabe5b

B0cho, 2025-06-07:
> @B0cho 
> 
> I'm duplicating the video from https://github.com/FreeCAD/FreeCAD/issues/18119 
> 
> https://github.com/user-attachments/assets/cc11ffa1-7865-4cdf-bc7c-1622dfaabe5b

@Pendelf thanks for reference. Then I think it is most reasonable go make it same as it is in Linkstage. I will try to do it.

hyarion, 2025-06-07:
> > What should be expected behaviour? Would it be, that this Dialog is opened a bit closer into screen center, no matter which panel it is located?
> 
> I think, screen center is the best option as you mentioned but we can open a discussion on the forum to hear other ideas and then we can decide easily
> 
> [image]

Screen center could be quite far away from the window. Maybe window center would be a better option?

B0cho, 2025-06-11:
Hi All,

What should be expected behavior, if `Model `panel is outside main window?
`Expression dialog` should always be moved into Main Window:

[image]

or should it hover `Model `panel as normally?

[image]

hyarion, 2025-06-11:
I just tested this on macOS and it seems like it is centered around the input field or button that opens the dialog.
My guess is that it tries to open the dialog on top of what is centered, which usually is a good idea as that's where you have your focus. This also minimizes the mouse movements.

Just thinking aloud here; how about just nudging the dialog so that no part of it is off screen?

semhustej, 2025-06-14:
> Hi everybody, Do I notice correctly, that under Linux `Expression dialog` is currently shown in window centre? Can someone confirm that? I would like to standarize position behavior on all platforms. [#18119 (comment)](https://github.com/FreeCAD/FreeCAD/issues/18119#issuecomment-2526205469)

Hi,

this is the default Expression editor location that I get in a **Virtual Machine**. I think the behavior is without and issue. The behavior is I think the one, that should be mimicked in Windows:

- the editor window is opened close to where user's mouse is
- if the window opens on the right side, it is aligned to the right side of the display

FreeCAD in fullscreen mode:

[image]

FreeCAD in non-fulscreen mode:

[image]

I am not sure, if VM has influence on the behavior.

```
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.42149 (Git) AppImage
Build date: 2025/06/13 08:44:01
Build type: Release
Branch: main
Hash: 9714e2f87a6240d1b2b979eb39c7d34791aafdae
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.958/1
Installed mods: 
  * Telemetry 1.0.0
```

ErtugrulEge, 2025-06-14:
What do you think about the center of the window for all operating systems? independent from property panel.

B0cho, 2025-06-14:
> What do you think about the center of the window for all operating systems? independent from property panel.

There was discussion (on discord if I remember correctly), that centre is not convenient as it requires to move mouse far from panel to screen centre.

semhustej, 2025-06-17:
That's quite strange indeed. Hopefully other Linux users can share the behavior on their machines.

hyarion, 2025-06-18:
@B0cho looking at the Qt version it looks like it is a Qt6 issue, as it worked for you when compiling with Qt5.

Could you try compiling with Qt6? (Can easily be tested without side effects with pixi in a separate worktree)

semhustej, 2025-08-16:
Tested with the latest dev version. Works like a charm. Thanks for your contribution!

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.2

```

PR #22426 (2025-08-07): Core: Fix 'Expression' and 'Vector' editor outhanging the screen
**Fix for the `Expression editor` and `Vector editor` outhanging screen when opened.**

### The Problem:
Previously, when these editors were opened from the _Model panel_ very close to the main window's edge, they could extend outside its boundaries, making user to move them inside.

[image]

### The Solution
 With this PR, the position of both editors will now automatically adjust if their initial opening position would cross the _main window's_ edge. This ensures they are always fully visible.

[image]

If the _Model panel_ is  [... cut]