FreeCAD/FreeCAD #21796: BIM: Roof does not auto-limit run length

Harvie, 2025-06-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Sometimes it happens that roof does not correctly end at the top, but continues further:

[image]

Please see this file:  [roof-intersect.FCStd.zip](https://github.com/user-attachments/files/20614022/roof-intersect.FCStd.zip)

### Full version info

```shell
OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.1.0dev.41837 (Git) AppImage
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Turntable/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 90/172.811/1
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-06-06:
I can confirm the the roof having the same shape as in the screenshot. If the base sketch is moved, the roof changes to be created correctly.

I don't know how to duplicate this behavior.

```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

semhustej, 2025-06-06:
This is how I can 'fix' it:

https://github.com/user-attachments/assets/13c10656-e659-43dc-8e63-56c80e3beb9d

closed 2025-06-06 by Roy-043

semhustej, 2025-06-06:
> Im wondering if there might be similar problem for roof lines that are not parallel. 

You might check that case once Roy's PR is released in the weekly.

PR #21814 (2025-06-06): BIM: Roof add tolerance to parallel check
Fixes #21796.
commit c567b75f (2025-06-06): BIM: Roof add tolerance to parallel check (#21814)

Fixes #21796.
commit a8bf9a5f (2025-07-11): BIM: Roof add tolerance to parallel check (#21814)

Fixes #21796.