FreeCAD/FreeCAD #21804: CAM: Tools - Editing enums not possible or very unfriendly

spanner888, 2025-06-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

pr #21425 Release Notes clearly state that editing should work in The New ToolBit Editor and be "more intuitive".

"Enhanced User Experience: Key UI Improvements
A primary focus of the btl branch was to improve the user interface for managing tools. Several key user interface components have been introduced or changed to provide a more intuitive and efficient workflow.

The New ToolBit Editor
The ToolBit Editor has been completely replaced. The new UI provides users with a clear visual representation of the tool, and allows to modify tool all properties - previously only dimensions were editable, now users can also edit multiple selection fields (enums) and switches (bool)."

I am unable to edit either of the default enums using the "Tool Lib Editor - Tool Parameter Editor", which appears to be what is referred to in above release notes.

Using the FreeCAD properties pane, I had some success and plenty of confusion.

Firstly user needs to know to expand the ToolCotroller, then in Properties Pane right click and enable show hidden.
Then to edit click on desired enum, then click on the three dots, then edit existing, or move to end of list and press enter to create a new line to enter the new enum. Press OK and new enum is available to use.

However if user tries to directly edit the enum list inplace, say by adding another, then yes new text is added in the property, but this is appended to name of existing enum. As you try to edit  the separating spaces and commas keep changing, ie spaces to commas and back again.

So adding an enum only works using the 3 dots "List Editor", but deleting an enum does work directly in the properties field.
 
All of this is very contrary to to oft repeated UI Improvements and  intuitive and efficient workflow.

Occurs on Win11 and Debian:

### Full version info

```shell
`OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/06/02 18:42:58
Build type: Release
Branch: main
Hash: 997d485124fab7b9ba971df59e91b55aeb5de61a
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/100.707/1
`

`OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/06/02 18:42:58
Build type: Release
Branch: main
Hash: 997d485124fab7b9ba971df59e91b55aeb5de61a
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.561/1.25`
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

spanner888, 2025-06-08:
There are two different attempts to edit enums in the above issue.
The first way tries to do what the Release Notes state is possible:
"The New ToolBit Editor
The ToolBit Editor has been completely replaced. The new UI provides users with a clear visual representation of the tool, and allows to modify tool all properties - previously only dimensions were editable, now users can also edit multiple selection fields (enums) and switches (bool)."

I am unable to find anyway to edit either enum:

[image]

Unless there is some hidden way to do this it is clearly a bug. If there is such a non-intuitive way to edit those enums using "The New ToolBit Editor", then it is still clearly a bug, as users will be unable to edit the enums.

So it is surprising that this issue has been marked as "Feature"

The second way mentioned was discovered while trying hard to find some other way than that documented above, namely through the properties pane.

[image]

[VIDEO link:](https://streamable.com/9hgxao) 

The video shows:
1. How it is easy to edit and mess up the enum.
2. That the list edit works. But it clearly needs at least some documentation to help users.
Again clearly a bug, as it does not meet the many claims of usability etc.

knipknap, 2025-06-08:
I see numerous topics in this issue, I will go through them one by one.

1. Editing **enum values** in the toolbit editor. I cannot see this in the screenshot, but it the enums are visible and appear editable, i.e. values can be chosen from the dropdown. So this part appears to work as intended.
2. Editing **enum ranges** in the tool dialog.
    - Editing the range of **built-in** attributes such as the spindle direction and tool materials is not intended. In fact, editing those risks breaking other things, as these builtin ranges are used by operations and other code to make decisions. Finding unexpected values in them can lead to problems elsewhere. It may also break during FreeCAD upgrades/data migrations.
    - Editing the range of **custom** attributes must be done within the shape file, as the properties (including enum range) are loaded from the shape, not defined in the toolbit. Editing them on the toolbit could also cause breakage because the tool becomes out-of-sync with the underlying shape.
3. Editing enum values in the property panel: This appears to work as intended in your video.
4. Editing enum ranges in the property panel: I agree this is not user friendly, but it is not CAM related. This is a separate issue, and improvements would need to be made in the core property editor. But in any case, circumventing protections to edit enum ranges can and in the future probably will lead to problems.

To summarize:
- Editing enum ranges anywhere except in the shape file is a bad idea and not intended to work. The tool material and spindle direction were made enums precisely to protect users from assigning custom values that could break things.
- Everything works as intended as far as I can see.

So unless I misunderstand something, IMO this issue should be closed.

*Side note: I think adding a "coating" value to the material enum is also logically inconsistent, as you can coat both, HSS and Carbide. It would be more logical to add a separate coating attribute.*

spanner888, 2025-06-09:
Regarding editing the enum:

1. What you describe as "edit" above, is considered just basic usage or changing of selected item in the selection list/dropdown. It is rare to even discuss enums in user facing documentation.

2. The release notes describe the ability to "edit" enums as a new feature "...previously only dimensions were editable, now users can also edit multiple selection fields (enums)..." .
This is incorrect as the existing Tool Library, Tool edit dialog already allowed users to change the ToolBit material using a drop down list.

3. Many users will misunderstand the use of "edit", especially with non-native English speakers reading the above and also after translation, where similar concepts often have subtle, but powerful distinctions in meaning.
The new Telemetry addon is showing that nearly 1/3 of users are using FreeCAD in another language. https://www.freecad.org/telemetry.php
There is no data on the number of non-native english speakers using FreeCAD with English language, but that use would very likely push the non-english use higher.

4. The easiest way to avoid confusing the users is some simple changes to the release notes.

5. All I have been doing is providing "feedback" based on information provided.
That feedback has not made any judgement on importance of issues reported, until point 3 above.
The testing undertaken has explicitly included testing of the release notes, as that is all the user facing documentation at this time.
Until point 4 above there has not been any suggestion, let alone instruction/direction of actions that might/should be taken.
This will be my last comment as I have "lost interest" in trying to help these changes succeed, so there is no need to write any response to me.

closed 2025-06-09 by maxwxyz