FreeCAD/FreeCAD #21805: CAM: can no longer create Profile or Facing Operation

spanner888, 2025-06-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Attempting to create a Profile or Facing Operation gives error below. 

`pyException: Traceback (most recent call last):
  File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-2025.06.02-Windows-x86_64-py311\Mod\CAM\Path\Op\Gui\Base.py", line 107, in setEdit
    self.setupTaskPanel(
  File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-2025.06.02-Windows-x86_64-py311\Mod\CAM\Path\Op\Gui\Base.py", line 120, in setupTaskPanel
    panel.setupUi()
  File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-2025.06.02-Windows-x86_64-py311\Mod\CAM\Path\Op\Gui\Base.py", line 1244, in setupUi
    page.addBaseGeometry(sel)
  File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-2025.06.02-Windows-x86_64-py311\Mod\CAM\Path\Op\Gui\Base.py", line 575, in addBaseGeometry
    if self.selectionSupportedAsBaseGeometry(selection, False):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-2025.06.02-Windows-x86_64-py311\Mod\CAM\Path\Op\Gui\Base.py", line 562, in selectionSupportedAsBaseGeometry
    if not self.supportsVertexes() and selection[0].SubObjects[0].ShapeType == "Vertex":
                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Base.FreeCADError'>: cannot determine type of null shape`

Issue occurs on Windows, but not on Debian.

To reproduce: 
Create new FC document, add a part cube. Then switch to CAM WB, select cube and create new Job, press OK to accept defaults.
Next Select cube top face and try to add either a Profile or Facing Op to the Job and error occurs.
Note have not tried other Ops.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/06/02 18:42:58
Build type: Release
Branch: main
Hash: 997d485124fab7b9ba971df59e91b55aeb5de61a
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.561/1.25
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-06:
@knipknap FYI

luzpaz, 2025-06-06:
> Version: 1.1.0dev.14555 (Git shallow)
> Hash: 997d485124fab7b9ba971df59e91b55aeb5de61a

Question: Do we derive the actual revision number from the hash and then add +14555 to it in order to know what revision is actually being tested ?

tarman3, 2025-06-06:
> [@tarman3](https://github.com/tarman3) could this be caused by your commit [b7ba562](https://github.com/FreeCAD/FreeCAD/commit/b7ba562288bcd168a3bfb084766799b260565bfa)?

Probably, No

@spanner888 
Did you try in Safe mode?

spanner888, 2025-06-07:
Version of FC is: [Development Build weekly-2025.06.02](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.06.02)

Config files where already first run, well maybe 2nd/3rd by time hit above issue, just created by FC an hour or so prior.

Safe mode makes no difference error still occurs.

tarman3, 2025-06-07:
Please try
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

spanner888, 2025-06-07:
That does not produce the above issue.
Thanks for your help.

spanner888, 2025-06-08:
Downloaded file: FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311.7z

FC Infos:
`OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.561/1.25`

spanner888, 2025-06-08:
In addition, just noticed below while opening AddonManager. Also occurs in safe mode.

No idea if related to initial report. Have not tried todo anything else.

`10:48:24  Loading PartDesign module... done
10:48:24  Loading GUI of PartDesign module... done
10:48:24  Init: Entering event loop
10:48:24  Init: Executing event loop...
10:48:25  Init: Processing command line files
10:48:25  Failed to lock file C:/Users/ours333/AppData/Local/cache/FreeCAD/Cache/FreeCAD_610747.lock
10:51:38  Checking network connection...
10:51:41  GitHub's zen message response: Non-blocking is better than blocking.
10:51:41  Traceback (most recent call last):
10:51:41    File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\AddonManager.py", line 253, in launch
10:51:41      metadata = MetadataReader.from_file(os.path.join(os.path.dirname(__file__), "package.xml"))
10:51:41                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
10:51:41    File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 261, in from_file
10:51:41      return MetadataReader.from_bytes(data)
10:51:41             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
10:51:41    File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 267, in from_bytes
10:51:41      return MetadataReader._process_element_tree(element_tree)
10:51:41             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
10:51:41    File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 273, in _process_element_tree
10:51:41      return MetadataReader._create_node(namespace, root)
10:51:41             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
10:51:41    File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 380, in _create_node
10:51:41      MetadataReader._parse_child_element(namespace, content_child, new_content_item)
10:51:41    File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 307, in _parse_child_element
10:51:41      metadata.license.append(MetadataReader._parse_license(child))
10:51:41                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
10:51:41    File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 326, in _parse_license
10:51:41      lm = get_license_manager()
10:51:41           ^^^^^^^^^^^^^^^^^^^^^
10:51:42    File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_licenses.py", line 174, in get_license_manager
10:51:42      _LICENSE_MANAGER = SPDXLicenseManager()
10:51:42                         ^^^^^^^^^^^^^^^^^^^^
10:51:42    File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_licenses.py", line 40, in __init__
10:51:42      self._load_license_data()
10:51:42    File "C:\Users\ours333\Documents\spanner\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_licenses.py", line 54, in _load_license_data
10:51:42      raise RuntimeError(f"File {spdx_path} not found")
10:51:42  RuntimeError: File C:\Users\ours333\Documents\spanner\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\./Resources/licenses/spdx.json not found`

spanner888, 2025-06-14:
Have just downloaded latest weekly https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
FreeCAD_weekly-builds-42149-conda-Windows-x86_64-py311.7z

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.42149 (Git)
Build date: 2025/06/13 08:44:01
Build type: Release
Branch: main
Hash: 9714e2f87a6240d1b2b979eb39c7d34791aafdae
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.561/1.25

and neither of the above issues occurs.

Of course another issue occurred, but that is most likely related to all the CAM WB changes, but it is now painfully obvious that it is pointless me wasting time and getting aggravated trying to contribute to CAM.