FreeCAD/FreeCAD #21810: CAM: Tool Library Editor - Default location of Add Existing file location dialog confusing.

spanner888, 2025-06-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

While attempting to use the CAM-Tool Library Editor - Add Existing, the file browser dialog opened in FC bin (ie executable) dir under W11.

On Debian the add existing opened to dir of last FreeCAD file opened, which also has no relation to location of *.fctb files.

Given new location of ToolBits etc and the differing locations on different platforms, this will cause ongoing confusion and more likely many users will be unable to add an existing toolbit.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/06/02 18:42:58
Build type: Release
Branch: main
Hash: 997d485124fab7b9ba971df59e91b55aeb5de61a
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.561/1.25

OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/06/02 18:42:58
Build type: Release
Branch: main
Hash: 997d485124fab7b9ba971df59e91b55aeb5de61a
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/100.707/1
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Harvie, 2025-06-06:
I agree that i was overall confused by tool management in couple weeks ago, when i started exploring FreeCAD CAM.
Especially the "toolbit dock" seems to be bit redundant. At first i thought i am supposed to use it to add tools to job. Only days later i've realized i can just doubleclick on job and switch to tools tab.

knipknap, 2025-06-06:
This issue is obsolete after #21769: There adding tools to a library is done directly in the UI by copying them from "all tools" to the specific library, so no file selector is needed for this purpose.

File selectors still exist for "import", and those open in the home dir of the user.

closed 2025-09-03 by sliptonic