FreeCAD/FreeCAD #21815: Text on Task panel cropped horizontally

furgo16, 2025-06-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I've noticed this using the BIM workbench, I'm not sure if it's a recent issue.

Steps to reproduce:

1. Start FreeCAD
2. From the Start page, click on the button to create an architectural project
3. Watch the task panel produce cropped text. A portion of the bottom part of the texts there is missing:

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42089 (Git) Snap 1482
Build date: 2025/06/05 17:18:41
Build type: Release
Branch: main
Hash: 4b068f4b8157c61560a42adf861b9b9a395742fa
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.1083/1
Installed mods: 
  * OpenTheme 2024.9.1
  * Movie 2025.1.4
  * sheetmetal 0.7.11
  * fasteners 0.5.38
  * fcVM
  * ezydxf
  * Render 2024.12.15
  * Arch
  * Road 2025.2.22
  * Assembly4 0.50.18
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-06-06:
@FEA-eng thanks for testing. I'm still on Qt 5, so this might not be Qt-specific.

semhustej, 2025-06-06:
I am able to reproduce only the task panel name text cutting, not the content of the task panel.

[image]
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

furgo16, 2025-06-16:
I can no longer reproduce it, thus closing the issue. Thanks for the heads up!

closed 2025-06-16 by furgo16