FreeCAD/FreeCAD #21829: Draft_Split moves the line that it's splitting

semhustej, 2025-06-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature**

**Current state**
Draft_Split not only splits the line, but sometimes it moves it unexpectedly.

_Case A:_

- open the file (remove the .zip extension)

[FC-split-tool.FCStd.zip](https://github.com/user-attachments/files/20632531/FC-split-tool.FCStd.zip)

- turn on `Draft_Snap_Intersection` and` Draft_Snap_WorkingPlane`
- start `Draft_Split`
- hover mouse over the bottom line
- than hover mouse over the top line, when the snap intersection point shows up, click
- the top line is split, but the line parts are also moved (only one end point, but it makes the tool do something different than split only)

See behavior in the video:

https://github.com/user-attachments/assets/b8afb680-8bbb-4401-b388-4bef1a412f9e

_Case B:_

The problematic split behavior happens when snaps are turned off. It is described in this video
https://youtu.be/TLY0voNJknU

**Proposed solution:**
`Draft_Split` tool should not do split+move. It should only split the existing geometry into two parts. The current behavior (especially case B) can lead to unexpected mistakes, which are hard to detect.
In the above described cases, the `Draft_Split` should leave the split ends of new lines in the position, where the original line was.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

Draft

### Anything else?

The above mentioned cases are different. I think it is more important to solve Case B, as that one can confuse new users very easily.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-06:
@Roy-043 FYI

closed 2025-08-05 by chennes

semhustej, 2025-08-16:
Tested with the latest dev version. Both cases described in the OP are fixed. Thanks a lot, this makes using the tool much more pleasant.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.2

```