FreeCAD/FreeCAD #21832: FreeCAD_1.0.1 Fusion of Polar Array & Sweep Fails

cropduster32, 2025-06-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I created an Archimedes blade using a helix and sketch, then offset the ruled surface by 2mm and made an array of 3 in draft. Made sure Fuse & Freenet were true. The created a sketch for the circular shaft, then used sweep to follow the length of the ruled surface. Applying a fusion to the sweep and the array cause the object to disappear and cannot be refreshed. The report view shows the following msg:

21:55:26  <Exception> TopoShapeExpansion.cpp(2484): no image for shape
22:09:38  pyException: Traceback (most recent call last):
  File "<string>", line 2, in <module>
<class 'ValueError'>: Null shape
22:09:38  Cannot convert Array because Null shape.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/Fusion
Installed mods: 
  * ThreadProfile 1.93.0
  * lattice2 1.0.0
  * Curves 0.6.61
  * CurvedShapes 1.0.13
  * Pyramids-and-Polyhedrons
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-06-07:
Hi @cropduster32 , could you add a minimum working example file that contains the error? It would be easier to confirm the issue from a file.

Katt-Kushol, 2025-06-08:
The reported behavior (fusing the array and the sweep failed) is confirmed.

Further investigation reveals, the array fails BOP check. In the tree under Array,  the first component fails BOP check is Offset. This  invalid geometry is propagating up the chain to affect the fusion process. 

[image]

The root cause maybe how the Ruled_Surface underneath the Offset is built. Worth taking a look into that code. 

Although a workaround is not the goal of this issue reporting, following workaround may indicate a way into the root cause. I modified Sketch under Rule_Surface: relocated the vertical line in there to introduce a small offset from the center of rotation. This allowed Offset and eventually the Array to pass BOP check and led to a successful Fusion. It almost feels like a surface between **a vertical line on the center** and the B-Spline was not forming well. File attached.

[Archimedes2_FR.zip](https://github.com/user-attachments/files/20642503/Archimedes2_FR.zip)

Version History:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41789 (Git)
Build date: 2025/05/19 18:24:17
Build type: Release
Branch: main
Hash: 1cfb85a71f4cd113deeccdc175850a37f2ad779d
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.9548/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * sheetmetal 0.7.22
  * Silk 0.1.7

maxwxyz, 2025-06-08:
@pieterhijma could this be OCC related?