FreeCAD/FreeCAD #21834: Draft: unexpected behavior when drafting in a view that is parallel to the working plane

semhustej, 2025-06-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open the file (remove the zip file extension)

[FC-draft-from-side.FCStd.zip](https://github.com/user-attachments/files/20637836/FC-draft-from-side.FCStd.zip)

- make sure snaps are turned off
- set view to `Front`
- start `Draft_Rectangle` or any other Draft tools
- click first point above the `Top` working plane
- click second point above the `Top` working plane
- Rectangle is created at an unexpected location very far from the origin
- This probably leads to graphical artefacts (faces are rendered in a corrupted way)

See behavior in the video:

https://github.com/user-attachments/assets/fbd50e7e-3803-4d83-84d0-238eb5170359

I can reproduce the issue in FreeCAD 1.0.1

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

### Subproject(s) affected?

Draft

### Anything else?

I am not sure what the behavior in this case should be. But the current behavior is confusing and should be handled in a different way in my opinion. Users should not be allowed to create 'corrupted' geometry so easily, even if it is their own mistake (I think this mistake can happen quite easily).

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-07:
Maybe related: https://github.com/FreeCAD/FreeCAD/issues/13537

ErtugrulEge, 2026-10-05:
I have changed tihs behaviour with the jelp of Claude AI. Code aligns view and working plane when they are parallel. AI add those methods to gui_snapper.py also I deleted default _get_wp and getApparentPoint  functions.

def _get_wp(self):
def is_view_parallel_to_wp(self, threshold=0.02):
def align_view_to_wp(self):
def getApparentPoint(self, x, y):  

Now when view is paraller you will se a warning message and FC aligns view according to wroking plane. 

@semhustej please test the file.Change the file Mod->Draft->Draftguitools->gui_snapper.py

Please test the code if the approach is correct we can check code quaility and make a pull request

[gui_snapper.py](https://github.com/user-attachments/files/33045274/gui_snapper.py)

https://github.com/user-attachments/assets/435083f3-8268-4fd5-8814-deaba8996fb8

semhustej, 2026-10-06:
@ErtugrulEge thanks for looking into the issue. I have not tested it yet, but I am not sure this is the right way to go. I think rotating the view could easily confuse users, as it is not consistent with how draft tools normally work. It could have unforseen consequences. There might be a case when user would want to snap the first point looking at the project parallel to working plane.

I need to think about this a bit more.