FreeCAD/FreeCAD #21835: CAM: Having two ways of managing job tools is confusing for newcomers

Harvie, 2025-06-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I think these two should get merged. It confused me as a newcomer that there are two redundant ways of adding bits:

[image]

And don't get me wrong. I prefer the way the right one is docked. But i don't like the idea of opening that window through anything else than job edit... Because when i first started with freecad cam, i didn't even knew i can add tools via job edit and this thing only gave me illusion that im on the right track doing it in some other ways...

### Full version info

```shell
OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/06/02 18:42:58
Build type: Release
Branch: main
Hash: 997d485124fab7b9ba971df59e91b55aeb5de61a
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Turntable/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 90/172.811/1
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

knipknap, 2025-06-07:
Personally, I never use the Job dialog way of adding tools, in fact I only discovered it when working on the CAM workbench source code recently 😄 . But I generally avoid the Job dialog because I know it causes recalculations, and I mostly work on large projects where that causes annoyingly long delays.

In other words, I personally wouldn't miss if the tool tab and dialog were removed from the job dialog entirely.

But I think this should be looked at by @sliptonic, as he added those dialogs AFAIK.

Harvie, 2025-06-07:
Yeah, there are pros and cons to both. That's why im suggesting "merging them" rather than "removing one".

sliptonic, 2025-06-07:
It's been many moons so my memory may be failing me.  IIRC, we added the panel in the job dialog expecting that we could eventually get rid of having individual document objects for TCs/tools in the tree.  That would clean up the tree but would mean the user needs a way to access tools&controllers already added.  @mlampert might have a better memory than I.

That said, I don't see us getting rid of those document objects any time soon.  Also, I agree that opening and closing the job dialog is too heavy and triggers too many recomputes. 

I'm cool with removing the panel from the job dialog.
Pinging @FreeCAD/cad-working-group to also weigh in.   If somebody feels up to it, posting a forum topic to invite more discussion before yeeting the panel would be friendly.

sliptonic, 2025-06-07:
I'm not sure the number of recomputes triggered by opening/closing the job is inappropriate.  It's just too many if you want to modify, for example a tool controller.

Harvie, 2025-06-27:
Off-topic rant on FreeCAD UX: #22206

sliptonic, 2025-06-27:
> Yeah, but the current workflow with the CAM Job dialog is that you sort of go through the setup of your CAM job and set options when you create a new job. Removing the tools from that workflow would mean you set up your job otherwise (stock etc) and then separately go and add tools.

That's one possible workflow.  I kind of regret that the job document object has grown into such a huge monster super task panel.  I mean you do SO much there that it's overwhelming.
- set up the job orientation and placement
- set up tools
- export templates
- configure default values
- set postprocessor
- configure postprocessor  options
- set up stock configuration 
- choose offset fixtures
- reorder operations

A lot of these things make sense and others  have accumulated over time.  I worry that the job dialog gets more and more confusing over time.

Connor9220, 2025-06-29:
While having things accessed from two different locations isn't ideal in most cases.  Having it here addresses two distinct workflows. 

If you know the tools your going to be using, you set those up straight away in the job editor dialog and just cam your part...
Additionally, the one in the job editor provides a complete list of all the tools in the job, feeds & speeds and is a one stop place to change those values. 

On the other hand, If you do not know what tools your going to use while setting up a job, well, that's when the tool dock comes in handy.  You don't have to go into the job and add them there. You get a human readable list of the tools. It's just a short cut - However, this is only to add tools, not edit existing ones, you still have to go through the job editor, OR edit the tool control directly int he Tree.   My issue with the tool dock is.. I only need it on occasion, so, I have to toggle it via keyboard commands (which I can never remember) or toggle it via Cam --> Toolbit Dock and then toggle it off when I'm done. (Because, it takes up valuable real-estate for something that's only used a few times)