FreeCAD/FreeCAD #21838: Regression: Closed Draft Wire cannot be dragged into a Body anymore

kwahoo2, 2025-06-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi,

This seems to be a recent regression. To reproduce:
1. Create a closed Wire in Draft, with "Make Face" enabled
2. Create a Body in Part Design
3. Drag the Wire into the Body (using the tree view).

Expected behaviour (how it works in 1.0.1 or earlier weeklies, eg. 41628):
Wire is (directly without BaseFeature) nested in the Body, can be used for additive features

What happens:
FreeCAD tries to create a BaseFeature, and fails with:
```
15:38:47  <Exception> TopoShapeExpansion.cpp(3885): No shells or compsolids found in shape
15:38:47  BaseFeature: No shells or compsolids found in shape
15:38:47  <Exception> TopoShapeExpansion.cpp(3885): No shells or compsolids found in shape
15:38:47  BaseFeature: No shells or compsolids found in shape
```
Interestingly only Wire fails, eg. Draft Polygon nests fine.

### Full version info

```shell
OS: Fedora Linux 42 (KDE Plasma Desktop Edition) (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.42095 (Git) AppImage
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/1
Installed mods: 
  * OpticsWorkbench 1.0.26
  * fasteners 0.5.35
  * sheetmetal 0.6.13
  * Curves 0.6.53
  * freecad.gears 1.3.0
  * freecad-xr-workbench 0.9.7
  * kicadStepUpMod 11.4.4
```

### Subproject(s) affected?

PartDesign

### Anything else?

OS: Fedora Linux 42 (KDE Plasma Desktop Edition) (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.42095 (Git) AppImage
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/1
Installed mods: 
  * OpticsWorkbench 1.0.26
  * fasteners 0.5.35
  * sheetmetal 0.6.13
  * Curves 0.6.53
  * freecad.gears 1.3.0
  * freecad-xr-workbench 0.9.7
  * kicadStepUpMod 11.4.4

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kwahoo2, 2025-06-09:
We should have a way do differentiate a Wire that have a face (all points coplanar, no self-intersection, area > 0).  Maybe there could be an one way conversion to Part2DObject with a coplanarity lock? Presumably automatic (opt-out), to avoid breaking people's old scripts.

Even now, if a ShapeBinder (created from a Draft Wire) is used for a Pad, it can be broken easily. Just move one of the Wire points out of the plane.

ZefengWang, 2025-12-02:
the same problem
export svg of paste layer from kicad
import the svg in freecad
create a body and place the objects in the group of body
it shows the error 
```
<Exception> TopoShapeExpansion.cpp(3844): No shells or compsolids found in shape
BaseFeature: No shells or compsolids found in shape
```