FreeCAD/FreeCAD #21839: PartDesign: Add the ability to create a centered cube

Pendelf, 2025-06-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Sometimes it is more convenient to create a cube in the center (Z-axis in the center of the cube), other figures are created in this way. It may be worth adding the option of placing the cube in the center 

[image]

https://github.com/user-attachments/assets/9e96f0a3-b5eb-499a-9559-60b3567ccb22

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * AirPlaneDesign 0.4.1
  * btl 0.9.9
  * CAM
  * CurvedShapes 1.0.13
  * Curves 0.6.63
  * fasteners 0.5.38
  * PieMenu 1.12.0
  * Road 2025.2.22
  * sheetmetal 0.7.22
  * Telemetry 1.0.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-07:
not sure if a separate command is needed? @FreeCAD/cad-working-group @FreeCAD/design-working-group

MisterMakerNL, 2025-06-07:
Could be useful if you want to mirror some features around the cube.
But also why not just make a sketch with a square, for that one time you need this?
Now I have to say I don't use any of the additive and subtractive shapes since I am used to making the sketch myself, so I am biased to this.
Basically other CAD programs don't have this, so my workflow doesn't have it and I don't use it.
I do understand why other programs don't have this, because it unnecessarily complicates things in my opinion.
I wonder if others use it too. maybe telemetry can say this?

pierreporte, 2025-06-07:
Definitely not a separate command, but rather a parameter.

kaiwas, 2025-06-08:
I can only add that primitives are good (especially cube). I also think that the issue of aligning objects also needs to be addressed and not only in PD. This is missing in PART or even BIM when creating beams and columns.

You just need the "offset" parameter with 9 points.