FreeCAD/FreeCAD #21841: FreeCAD_weekly-builds-42095-conda-Windows - Addon Manager not functional

Larry0815, 2025-06-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I start Addon manager it gives me that:

18:51:26  Traceback (most recent call last):
18:51:26    File "D:\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\AddonManager.py", line 253, in launch
18:51:26      metadata = MetadataReader.from_file(os.path.join(os.path.dirname(__file__), "package.xml"))
18:51:26                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
18:51:26    File "D:\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 261, in from_file
18:51:26      return MetadataReader.from_bytes(data)
18:51:26             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
18:51:26    File "D:\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 267, in from_bytes
18:51:26      return MetadataReader._process_element_tree(element_tree)
18:51:26             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
18:51:26    File "D:\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 273, in _process_element_tree
18:51:26      return MetadataReader._create_node(namespace, root)
18:51:26             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
18:51:26    File "D:\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 380, in _create_node
18:51:26      MetadataReader._parse_child_element(namespace, content_child, new_content_item)
18:51:26    File "D:\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 307, in _parse_child_element
18:51:26      metadata.license.append(MetadataReader._parse_license(child))
18:51:26                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
18:51:26    File "D:\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_metadata.py", line 326, in _parse_license
18:51:26      lm = get_license_manager()
18:51:26           ^^^^^^^^^^^^^^^^^^^^^
18:51:26    File "D:\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_licenses.py", line 174, in get_license_manager
18:51:26      _LICENSE_MANAGER = SPDXLicenseManager()
18:51:26                         ^^^^^^^^^^^^^^^^^^^^
18:51:26    File "D:\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_licenses.py", line 40, in __init__
18:51:26      self._load_license_data()
18:51:26    File "D:\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\.\addonmanager_licenses.py", line 54, in _load_license_data
18:51:26      raise RuntimeError(f"File {spdx_path} not found")
18:51:26  RuntimeError: File D:\FreeCAD_weekly-builds-42095-conda-Windows-x86_64-py311\Mod\AddonManager\./Resources/licenses/spdx.json not found

weekly-builds-42006- works as expected .

Best regards , Larry

### Full version info

```shell
OS: Windows 10 build 19044
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/54.1867/1.25
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-06-07:
I can confirm

Pendelf, 2025-06-07:
I also confirm

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * AirPlaneDesign 0.4.1
  * btl 0.9.9
  * CAM
  * CurvedShapes 1.0.13
  * Curves 0.6.63
  * fasteners 0.5.38
  * PieMenu 1.12.0
  * Road 2025.2.22
  * sheetmetal 0.7.22
  * Telemetry 1.0.1

maxwxyz, 2025-06-07:
@oursland FYI, maybe because of submodule?

furgo16, 2025-06-08:
Duplicate of https://github.com/FreeCAD/AddonManager/issues/100

maxwxyz, 2025-06-08:
so fixed by https://github.com/FreeCAD/AddonManager/pull/101

closed 2025-06-08 by maxwxyz