FreeCAD/FreeCAD #21842: FreeCAD 1.0.1 Crashing on Linux aarch64 when saving

RiderExMachina, 2025-06-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This may be related to #16942.

I'm running the Flatpak build of FreeCAD on Asahi Fedora 42.
```
Operating System: Fedora Linux Asahi Remix 42
KDE Plasma Version: 6.3.5
KDE Frameworks Version: 6.14.0
Qt Version: 6.9.0
Kernel Version: 6.14.6-400.asahi.fc42.aarch64+16k (64-bit)
Graphics Platform: Wayland
Processors: 4 × Apple Firestorm (M1), 4 × Apple Icestorm (M1)
Memory: 7.3 GiB of RAM
Graphics Processor 1: Apple M1
Graphics Processor 2: Apple M1
Graphics Processor 3: Apple M1
Product Name: Apple MacBook Pro (13-inch, M1, 2020)
```

The program is usable and performant up until I need to save, at which point FreeCAD crashes with a Segmentation Error. This happens no matter if I try `ctrl+S` or using `File > Save As...` Below is the output when I load FreeCAD, recover a sketch, and try to save.

```
> flatpak run org.freecad.FreeCAD

FreeCAD 1.0.1, Libs: 1.0.1R39285 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Qt: Session management error: Could not open network socket
<ElementMap> ElementMap.cpp(476): No hasherRef
Reading failed from embedded file: Sketch.InternalShape.bin (0 bytes, 2 bytes compressed)
<ElementMap> ElementMap.cpp(476): No hasherRef
<ElementMap> ElementMap.cpp(476): No hasherRef
Reading failed from embedded file: Pad.SuppressedShape.bin (0 bytes, 2 bytes compressed)
Reading failed from embedded file: Sketch001.InternalShape.bin (0 bytes, 2 bytes compressed)
Reading failed from embedded file: Pocket.AddSubShape.bin (0 bytes, 2 bytes compressed)
Reading failed from embedded file: Pocket.Shape.bin (0 bytes, 2 bytes compressed)
Reading failed from embedded file: Pocket.SuppressedShape.bin (0 bytes, 2 bytes compressed)
Program received signal SIGSEGV, Segmentation fault.
#0  linux-vdso.so.1(__kernel_rt_sigreturn+0) [0xffff341e8840]
#1  /usr/lib/aarch64-linux-gnu/libQt6Gui.so.6(+0x1a777c) [0xffff2fb4777c]
#2  0xffff2f4f8118 in QVariant::~QVariant() from /usr/lib/aarch64-linux-gnu/libQt6Core.so.6+0x90
#3  0xffff307eb4b4 in QHeaderView::sectionSizeFromContents(int) const from /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6+0xcc
#4  0xffff307e95b4 in QHeaderView::sizeHint() const from /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6+0x8c
#5  0xffff30866318 in QTreeView::updateGeometries() from /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6+0x158
#6  0xffff307ca5fc in QAbstractItemView::doItemsLayout() from /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6+0x30
#7  0xffff308645ac in QTreeView::doItemsLayout() from /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6+0x80
#8  /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6(+0x446484) [0xffff30776484]
#9  0xffff30776ad4 in QFileDialog::setOptions(QFlags<QFileDialog::Option>) from /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6+0x134
#10  0xffff32818a70 in Gui::FileDialog::getSaveFileName(QWidget*, QString const&, QString const&, QString const&, QString*, QFlags<QFileDialog::Option>) from /app/freecad/lib/libFreeCADGui.so+0x16c
#11  0xffff32262994 in Gui::Document::saveAs() from /app/freecad/lib/libFreeCADGui.so+0x248
#12  0xffff32714cd0 in Gui::View3DInventor::onMsg(char const*, char const**) from /app/freecad/lib/libFreeCADGui.so+0x490
#13  0xffff321dd6a4 in Gui::Application::sendMsgToActiveView(char const*, char const**) from /app/freecad/lib/libFreeCADGui.so+0x3c
#14  0xffff322200f8 in Gui::Application::sSendActiveView(_object*, _object*) from /app/freecad/lib/libFreeCADGui.so+0x6c
#15  /usr/lib/aarch64-linux-gnu/libpython3.12.so.1.0(+0x1ee3e4) [0xffff30d6e3e4]
#16  /usr/lib/aarch64-linux-gnu/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x84) [0xffff30cdf370]
#17  /usr/lib/aarch64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x490c) [0xffff30ec27ac]
#18  /usr/lib/aarch64-linux-gnu/libpython3.12.so.1.0(PyEval_EvalCode+0xd4) [0xffff30ecd9b4]
#19  /usr/lib/aarch64-linux-gnu/libpython3.12.so.1.0(+0x34e5a8) [0xffff30ece5a8]
#20  /usr/lib/aarch64-linux-gnu/libpython3.12.so.1.0(+0x34e6a4) [0xffff30ece6a4]
#21  /usr/lib/aarch64-linux-gnu/libpython3.12.so.1.0(PyRun_StringFlags+0x80) [0xffff30ece7a8]
#22  0xffff313b14e4 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /app/freecad/lib/libFreeCADBase.so+0x84
#23  0xffff32333d88 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /app/freecad/lib/libFreeCADGui.so+0xa4
#24  0xffff32333f50 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /app/freecad/lib/libFreeCADGui.so+0xc8
#25  0xffff32339160 in Gui::Command::_invoke(int, bool) from /app/freecad/lib/libFreeCADGui.so+0x2dc
#26  0xffff32339898 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /app/freecad/lib/libFreeCADGui.so+0x118
#27  /usr/lib/aarch64-linux-gnu/libQt6Core.so.6(+0x1ff080) [0xffff2f4df080]
#28  0xffff2ffbe1ec in QAction::triggered(bool) from /usr/lib/aarch64-linux-gnu/libQt6Gui.so.6+0x40
#29  0xffff2ffc1150 in QAction::activate(QAction::ActionEvent) from /usr/lib/aarch64-linux-gnu/libQt6Gui.so.6+0xe8
#30  /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6(+0x3956d8) [0xffff306c56d8]
#31  /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6(+0x39db88) [0xffff306cdb88]
#32  0xffff30550c7c in QWidget::event(QEvent*) from /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6+0x1dc
#33  0xffff304eec0c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6+0x5c
#34  0xffff304f7ba4 in QApplication::notify(QObject*, QEvent*) from /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6+0x304
#35  0xffff322d82d8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0xec
#36  0xffff2f472434 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/aarch64-linux-gnu/libQt6Core.so.6+0x174
#37  0xffff304f6e9c in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6+0x1b4
#38  /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6(+0x234d04) [0xffff30564d04]
#39  /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6(+0x236f5c) [0xffff30566f5c]
#40  0xffff304eec0c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/aarch64-linux-gnu/libQt6Widgets.so.6+0x5c
#41  0xffff322d82d8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0xec
#42  0xffff2f472434 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/aarch64-linux-gnu/libQt6Core.so.6+0x174
#43  0xffff2fbad35c in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/aarch64-linux-gnu/libQt6Gui.so.6+0x314
#44  0xffff2fc1aaf8 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/aarch64-linux-gnu/libQt6Gui.so.6+0xd0
#45  /usr/lib/aarch64-linux-gnu/libQt6XcbQpa.so.6(+0x5d788) [0xffff2911d788]
#46  /usr/lib/aarch64-linux-gnu/libglib-2.0.so.0(+0x61d90) [0xffff2d3e1d90]
#47  /usr/lib/aarch64-linux-gnu/libglib-2.0.so.0(+0xc1278) [0xffff2d441278]
#48  /usr/lib/aarch64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x34) [0xffff2d3e10d8]
#49  0xffff2f77c3e4 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/aarch64-linux-gnu/libQt6Core.so.6+0x54
#50  0xffff2f47f884 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/aarch64-linux-gnu/libQt6Core.so.6+0x194
#51  0xffff2f47ab90 in QCoreApplication::exec() from /usr/lib/aarch64-linux-gnu/libQt6Core.so.6+0xa0
#52  0xffff321ef650 in Gui::Application::runApplication() from /app/freecad/lib/libFreeCADGui.so+0xa84
#53  FreeCAD(+0x72f4) [0xaaab40fe72f4]
#54  /usr/lib/aarch64-linux-gnu/libc.so.6(+0x261d4) [0xffff2ee461d4]
#55  /usr/lib/aarch64-linux-gnu/libc.so.6(__libc_start_main+0x98) [0xffff2ee462b8]
#56  FreeCAD(+0x8270) [0xaaab40fe8270]
```

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: arm64
Version: 1.0.1.39285 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 878f0b8)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Anything else?

My next step will be to try the development branch to see if there are any changes since 1.0.1 and I will report back.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-08:
I have no access to the repo, @adrianinsaval maybe?

luzpaz, 2025-07-28:
@adrianinsaval transfer to flatpak repo please

closed 2026-02-07 by RiderExMachina

Evgsurf, 2026-09-14:
I got today. Yesterday it was Okey. (I guess ubuntu was updated by nigth)
I can not start FreeCAD.

Ubuntu 22.04. Freecad 1.1.3, Libs: 1.1.3R44987 (Git) from Flathab 

```
$flatpak run org.freecad.FreeCAD

FreeCAD 1.1.3, Libs: 1.1.3R44987 (Git)
(C) 2001-2011 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Qt: Session management error: Could not open network socket
During initialization the error "/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API" occurred in /app/freecad/Mod/Assembly/InitGui.py
Look into the log file for further information
During initialization the error "/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API" occurred in /app/freecad/Mod/Measure/InitGui.py
Look into the log file for further information
During initialization the error "/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API" occurred in /app/freecad/Mod/Draft/InitGui.py
Look into the log file for further information
During initialization the error "/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API" occurred in /app/freecad/Mod/BIM/InitGui.py
Look into the log file for further information
During initialization the error "No viable version of PySide was found (tried the FreeCAD PySide wrapper, PySide6 and PySide2)" occurred in /app/freecad/Mod/AddonManager/./InitGui.py
Look into the log file for further information
During initialization the error "/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API" occurred in /app/freecad/Mod/Tux/InitGui.py
Look into the log file for further information
During initialization the error "/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API" occurred in /app/freecad/Mod/Help/InitGui.py
Look into the log file for further information
Wizard shaft module cannot be loaded
/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API
Wizard shaft module cannot be loaded
/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API
'BasicShapes' package cannot be loaded. libshiboken: could not import module 'PySide6.QtCore'
'CompoundTools' package cannot be loaded. libshiboken: could not import module 'PySide6.QtCore'
'BOPTools' package cannot be loaded. /app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API
Unknown command 'Part_Tube'
Unknown command 'Part_Tube'
Unknown command 'Part_JoinConnect'
Unknown command 'Part_JoinEmbed'
Unknown command 'Part_JoinCutout'
Unknown command 'Part_BooleanFragments'
Unknown command 'Part_SliceApart'
Unknown command 'Part_Slice'
Unknown command 'Part_XOR'
Unknown command 'Part_ExplodeCompound'
Unknown command 'Part_CompoundFilter'
Unknown command 'Part_ToleranceSet'
Error in Profiles module
/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_APIError: Initializing one or more of the Draft modules failed, Draft will not work as expected.
/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API
/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API
/app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/app/freecad/Mod/MeshPart/MeshFlatteningCommand.py", line 31, in <module>
    from PySide.QtCore import QT_TRANSLATE_NOOP  # for translations
  File "/app/freecad/Ext/PySide/QtCore.py", line 1, in <module>
    from PySide6.QtCore import *
<class 'ImportError'>: /app/lib/python3.13/site-packages/PySide6/QtCore.cpython-313-x86_64-linux-gnu.so: undefined symbol: _ZN14QObjectPrivateC2E16QtPrivate_6_11_1, version Qt_6_PRIVATE_API

```