FreeCAD/FreeCAD #21848: BIM: hexagonal roof run length not limited properly

Harvie, 2025-06-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I was wondering about this since encountering #21796
However it works correctly when correctly truncated run length is entered by user. Im not sure if it's worth the hassle to fix this, because i don't really live in hexagonal building :-) Just reporting it in case you wanna know, close it if you consider irrelevant.

[image]

[roof-hex-bug.FCStd.zip](https://github.com/user-attachments/files/20640754/roof-hex-bug.FCStd.zip)

---------------

If on the other hand, you feel adventurous, i can come up with more stress tests 🤣 

[image]

### Full version info

```shell
Also verified in 1.1.0 release

OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/06/02 18:42:58
Build type: Release
Branch: main
Hash: 997d485124fab7b9ba971df59e91b55aeb5de61a
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Turntable/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 90/172.811/1
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Harvie, 2025-06-07:
Haha, i'm actually impressed with this one...

[image]

Im not even sad i had to enter the run lengths manualy...

Roy-043, 2025-06-08:
The algorithm compares roof segments that come before and after another segment. It does not handle this layout.

semhustej, 2025-10-15:
> I only see the sketch when I open the file attached to the description of this issue. The Roof object is present and has visibility enabled but there's no effect in the 3D view.
> 

Run0 has to be changed from 1000cm to something smaller (100cm in my case) for the roof to be visible.

[image]

semhustej, 2026-01-14:
The issue is still valid.