FreeCAD/FreeCAD #21861: Segfault when removing multiple objects at the same time

Rexbas, 2025-06-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

See the attached file. It is a modified version of the file on the forum here: https://forum.freecad.org/viewtopic.php?t=97208.

When selecting the last 3 objects in the tree view, like in the image below, and pressing delete, then yes in the popup. FreeCAD crashes with a segfault. Removing the objects one by one does not trigger the segfault.

[image]

<details>
  <summary>Stack trace</summary>
  
  ```
 Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x19c30) [0x7f3e64b57c30]
#1  0x7f3e6b66b49b in Gui::DocumentObjectItem::testStatus(bool, QIcon&, QIcon&) from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0xadb
#2  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x26674d5) [0x7f3e6b6674d5]
#3  0x7f3e6b65ae49 in Gui::DocumentItem::testStatus() from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0x79
#4  0x7f3e6b63ce2f in Gui::TreeWidget::onUpdateStatus() from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0xd3f
#5  0x7f3e6b640d1f in Gui::TreeWidget::_updateStatus(bool) from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0x7f
#6  0x7f3e6b63eee9 in Gui::TreeWidget::onSelectTimer() from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0x29
#7  0x7f3e6b63e264 in Gui::TreeWidget::onItemSelectionChanged() from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0xe4
#8  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x26808ea) [0x7f3e6b6808ea]
#9  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x268083d) [0x7f3e6b68083d]
#10  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x2680767) [0x7f3e6b680767]
#11  /home/bas/src/FreeCAD/.pixi/envs/default/lib/libQt6Core.so.6(+0x30a527) [0x7f3e6530a527]
#12  /home/bas/src/FreeCAD/.pixi/envs/default/lib/libQt6Widgets.so.6(+0x1b9ac7) [0x7f3e661b9ac7]
#13  /home/bas/src/FreeCAD/.pixi/envs/default/lib/libQt6Core.so.6(+0x30a527) [0x7f3e6530a527]
#14  0x7f3e651252e1 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /home/bas/src/FreeCAD/.pixi/envs/default/lib/libQt6Core.so.6+0x41
#15  /home/bas/src/FreeCAD/.pixi/envs/default/lib/libQt6Core.so.6(+0x1d1195) [0x7f3e651d1195]
#16  /home/bas/src/FreeCAD/.pixi/envs/default/lib/libQt6Core.so.6(+0x30a527) [0x7f3e6530a527]
#17  0x7f3e65124bff in QAbstractItemModel::rowsAboutToBeInserted(QModelIndex const&, int, int, QAbstractItemModel::QPrivateSignal) from /home/bas/src/FreeCAD/.pixi/envs/default/lib/libQt6Core.so.6+0x5f
#18  0x7f3e651da610 in QAbstractItemModel::beginInsertRows(QModelIndex const&, int, int) from /home/bas/src/FreeCAD/.pixi/envs/default/lib/libQt6Core.so.6+0x70
#19  0x7f3e66248110 in QTreeWidgetItem::insertChild(int, QTreeWidgetItem*) from /home/bas/src/FreeCAD/.pixi/envs/default/lib/libQt6Widgets.so.6+0xf0
#20  0x7f3e6b659e12 in Gui::DocumentItem::createNewItem(Gui::ViewProviderDocumentObject const&, QTreeWidgetItem*, int, std::shared_ptr<Gui::DocumentObjectData>) from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0x412
#21  0x7f3e6b6658cc in Gui::TreeWidget::_slotDeleteObject(Gui::ViewProviderDocumentObject const&, Gui::DocumentItem*) from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0x7ac
#22  0x7f3e6b661de2 in Gui::TreeWidget::slotDeleteObject(Gui::ViewProviderDocumentObject const&) from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0x22
#23  0x7f3e6b691aea in void std::__invoke_impl<void, void (Gui::TreeWidget::*&)(Gui::ViewProviderDocumentObject const&), Gui::TreeWidget*&, Gui::ViewProviderDocumentObject const&>(std::__invoke_memfun_deref, void (Gui::TreeWidget::*&)(Gui::ViewProviderDocumentObject const&), Gui::TreeWidget*&, Gui::ViewProviderDocumentObject const&) from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0x6a
#24  0x7f3e6b691a55 in std::__invoke_result<void (Gui::TreeWidget::*&)(Gui::ViewProviderDocumentObject const&), Gui::TreeWidget*&, Gui::ViewProviderDocumentObject const&>::type std::__invoke<void (Gui::TreeWidget::*&)(Gui::ViewProviderDocumentObject const&), Gui::TreeWidget*&, Gui::ViewProviderDocumentObject const&>(void (Gui::TreeWidget::*&)(Gui::ViewProviderDocumentObject const&), Gui::TreeWidget*&, Gui::ViewProviderDocumentObject const&) from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0x25
#25  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x2691a0f) [0x7f3e6b691a0f]
#26  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x2691983) [0x7f3e6b691983]
#27  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x2691785) [0x7f3e6b691785]
#28  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x20ac128) [0x7f3e6b0ac128]
#29  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x20ac095) [0x7f3e6b0ac095]
#30  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x20ac039) [0x7f3e6b0ac039]
#31  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x20abfee) [0x7f3e6b0abfee]
#32  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x20abeb6) [0x7f3e6b0abeb6]
#33  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x20abe45) [0x7f3e6b0abe45]
#34  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x20abd8d) [0x7f3e6b0abd8d]
#35  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x20abcb8) [0x7f3e6b0abcb8]
#36  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x20aba71) [0x7f3e6b0aba71]
#37  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x20ab7a1) [0x7f3e6b0ab7a1]
#38  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x203a329) [0x7f3e6b03a329]
#39  0x7f3e6b13602a in Gui::Document::slotDeletedObject(App::DocumentObject const&) from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0x21a
#40  0x7f3e6b162eea in void std::__invoke_impl<void, void (Gui::Document::*&)(App::DocumentObject const&), Gui::Document*&, App::DocumentObject const&>(std::__invoke_memfun_deref, void (Gui::Document::*&)(App::DocumentObject const&), Gui::Document*&, App::DocumentObject const&) from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0x6a
#41  0x7f3e6b162e15 in std::__invoke_result<void (Gui::Document::*&)(App::DocumentObject const&), Gui::Document*&, App::DocumentObject const&>::type std::__invoke<void (Gui::Document::*&)(App::DocumentObject const&), Gui::Document*&, App::DocumentObject const&>(void (Gui::Document::*&)(App::DocumentObject const&), Gui::Document*&, App::DocumentObject const&) from /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so+0x25
#42  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x2162dcf) [0x7f3e6b162dcf]
#43  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x2162d43) [0x7f3e6b162d43]
#44  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADGui.so(+0x2162b45) [0x7f3e6b162b45]
#45  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADApp.so(+0x6b97b8) [0x7f3e68ab97b8]
#46  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADApp.so(+0x6b9725) [0x7f3e68ab9725]
#47  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADApp.so(+0x6b96c9) [0x7f3e68ab96c9]
#48  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADApp.so(+0x6b967e) [0x7f3e68ab967e]
#49  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADApp.so(+0x6b9546) [0x7f3e68ab9546]
#50  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADApp.so(+0x6b94d5) [0x7f3e68ab94d5]
#51  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADApp.so(+0x6b941d) [0x7f3e68ab941d]
#52  /home/bas/src/FreeCAD/build/debug/lib/libFreeCADApp.so(+0x6b9348) [0x7f3e68ab9348]
[... 61 lines cut ...]
```shell
OS: Fedora Linux 42 (KDE Plasma Desktop Edition) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.42101 (Git)
Build date: 2025/06/08 07:51:25
Build type: Unknown
Branch: main
Hash: e4b1512ae9eb91c26b0b4c65695c9f10dc5bb6c0
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
Installed mods: 
  * OpenTheme 2025.5.20
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-08:
maybe related?
- https://github.com/FreeCAD/FreeCAD/issues/20733
- https://github.com/FreeCAD/FreeCAD/issues/20304

maxwxyz, 2025-06-10:
I can confirm: no issues when selecting with control, order does not matter. If I select with Shift from bottom to top and delete it crashes. I assume as Shift also selects the nested items under the Pad/Cut,... ?

I'm also wondering: how is there a PartDesign Pad outside a body?

```
OS: macOS 13.7.6
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/72/1
Installed mods: 
  * freecad.gears 1.3.0
  * CfdOF 1.30.4
  * Defeaturing 1.2.6
  * Gridfinity 0.11.9
  * Manipulator 1.6.0
  * fasteners 0.5.38
  * Telemetry 1.0.1
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.22
  * FusedFilamentDesign 0.25.200
  * Curves 0.6.62

```

closed 2025-06-16 by kadet1090