FreeCAD/FreeCAD #21863: BIM: BIM spatial Objects (Level, Building) do not deactivate is certain scenario

semhustej, 2025-06-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open file (remove the .zip file extension)

[FC-levels-axctivation.FCStd.zip](https://github.com/user-attachments/files/20643554/FC-levels-axctivation.FCStd.zip)

- right click `Level 01` and select `Active` to activate the level
- right click `Level 03` and select `Active` to activate the level
- everything is all right
- right click `Level 03` and select `Active` to deactivate the level
- the result is that `Level 01` is activated. This is unexpected. The expected behavior is that no object should be active.

The same behavior happens when a building object is activated directly after another spatial object in `BIM_Views` panel.
I tested with `Part Design` Body objects and this unexpected behavior does not happen in `Part Design` workbench. It seems to be related to BIM objects.

See behavior in the video:

https://github.com/user-attachments/assets/6d95c0dd-b6b1-4a6b-96f8-6a26b13a7170

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tetektoza, 2025-06-08:
My regression most probably. Will check it today. BTW, @semhustej forgot if I asked you, would it be possible if you would join our discord too?

semhustej, 2025-06-08:
>  would it be possible if you would join our discord too?

Sure. How can I join? Do I need to be invited? I have not used discord yet.

tetektoza, 2025-06-08:
@semhustej here, does it work for you?: https://discord.com/invite/w2cTKGzccC

tetektoza, 2025-06-08:
@semhustej Deleted previous messages so bots won't cache our discord link and spam it.

Now looking at it it looks like correct behavior. I've made it so it works like a stack:
If you have selected:
Top WP (default) -> Level 1 -> Level 3, then unselecting Level 3 makes you go to Level 1 and then to Top WP (default). It basically pops up previous WP and goes into it. I think it worked like this before, no? I could be wrong. Please, if you could check how it worked for BIM, but I think I aligned it to something so it had reason.

kaiwas, 2025-06-09:
I also found it strange. Automatic activation of the previous remembered "level" is more of an inconvenience.
If there are a lot of levels and switching occurs randomly - we will simply get confused, because there is no identifier of which active level was the previous remembered one.

In addition, sometimes it is necessary to deactivate ALL levels to create an object at the bottom of the view tree.

Probably, it is necessary to simplify the approach. Make only two values ​​"Active"/"Inactive". This will be more predictable behavior.

tetektoza, 2025-06-09:
@semhustej doesn't it behave the same way on your demo? You activate level 1 -> you activate level 3 -> you deactivate level 3 (you are now on level 1). Looks the same, I think I haven't modified the stack behavior, just aligned to what it was

semhustej, 2025-06-09:
> [@semhustej](https://github.com/semhustej) doesn't it behave the same way on your demo? You activate level 1 -> you activate level 3 -> you deactivate level 3 (you are now on level 1). Looks the same, I think I haven't modified the stack behavior, just aligned to what it was

Note that the switcher in the top left corner of my monitor is _Active Workplane Switcher_. This is different than what _Active Container_ (Level) is. The Active Container is visible only thanks to the container being highlighted in green in the project tree. These are two independent settings which don't necessarily be linked together.
An example is in this screenshot:

[image]

semhustej, 2025-06-09:
I would do it this way:

> 
> Deactivate deactivates level completely, doesn't activate any other one, ~~and sets the primary default workplane that's set during object creation.~~

I would not manipulate the workplane during `Deactivate` command at all.

tetektoza, 2025-06-09:
> I would do it this way:
> 
> > Deactivate deactivates level completely, doesn't activate any other one, ~and sets the primary default workplane that's set during object creation.~
> 
> I would not manipulate the workplane during `Deactivate` command at all.
> 
> But this is a change compared to how the tool works in FC 1.0, that's why I wanted to know what others think about it before you make the change.

Activate is all about working plane manipulation so I'd expect from deactivation to just set Active Workplane to the default one, otherwise it's weird. Basically deactivate would do nothing?

kaiwas, 2025-06-09:
> Activate is all about working plane manipulation so I'd expect from deactivation to just set Active Workplane to the default one, otherwise it's weird. Basically deactivate would do nothing?

I don't mind that the work plane moves to the active floor when activated. It's really convenient.
When deactivated, the work plane could be placed in the initial state (0,0,0). IMHO

tetektoza, 2025-06-09:
Nevertheless I still think we should revert the working plane, if it has been an effect of Activate, then counter effect that I'd expect during Deactivate would be to remove/revert what Activate did. Also, this is behavior that was already shown in the PR on the demo, why we haven't discussed it back then and now we're going over this once again?

kaiwas, 2025-06-09:
> why we haven't discussed it back then and now we're going over this once again?

This is a subtle point. For example, I did not notice that when deactivating, something else is activated.

Okay. What is the way to remember what was activated previously and, most importantly, how to quickly reset all activations?

furgo16, 2025-06-09:
> When activated, the container is activated and the working plane is set to the container level.
> When deactivated, deactivation occurs without activating another level. The working plane is set to zero coordinates.

That seems to me as the simplest behaviour, and the least likely to surprise the user.

That means changing [current behaviour](https://wiki.freecad.org/Arch_BuildingPart#Options): 

> By double-clicking the BuildingPart object in the tree view, the [Working Plane](https://wiki.freecad.org/Draft_SelectPlane) will be set to its location, and the BuildingPart will become active, which means that new objects will be added automatically to it. Double-clicking the BuildingPart again will deactivate it and set the working plane back to its previous position

In other words, unless I interpret it incorrectly from reading the issue, activation/deactivation works as designed, and this is a change request.

I think the reason it wasn't mentioned before was that not many of us (myself included) were too familiar with the activation/deactivation mechanism, since it was disabled by default. We recently activated it by default, so it's been more scrutinized as people have started putting it into practice.

I'm in two minds about it: I like the simplicity of the new proposal, but I do wonder whether there is a use case for shelving the previous working plane upon activation and popping it upon deactivation. @yorikvanhavre what are your thoughts?

semhustej, 2025-06-09:
Also worth considering this scenario:

- user activates Level 01
- user changes working plane while the Level is activated (maybe user needs to create an objects using Front working plane). It is quite reasonable - user needs to create an object on Level 01, but the working plane needs to be different than the default one.
- now user deactivates Level 01. What does the working plane plane change to?

In FC 1.0 the `Deactivate` tool in this case does not change the working plane. In the above mentioned scenario it is still `Front` after the deactivation. This is good behavior in my opinion.

kaiwas, 2025-06-10:
I thought three more times ))) 
Probably the problem is not so critical for me. Perhaps it is possible to keep only active levels in the scene and never deactivate them.
Just switch to the desired floor when required.

So the current situation suits me quite well, although I think that it is not very correct.

semhustej, 2025-09-20:
@tetektoza will you be able to fix this before the 1.1 release, so that the new functionality is not half-baked in the new release?

semhustej, 2025-09-21:
**Let's pin this as the desired behavior:**

> 
>     * When activated, the container is activated and the working plane is set to the container level.
> 
>     * When deactivated, deactivation occurs without activating another level. The working plane is set to zero coordinates.

I would add that 'zero coordinates' means the default `Top (XY)` working plane.

The discussion got quite long, I think these two bullet points summarize the agreed desired outcome.

furgo16, 2025-09-21:
> The discussion got quite long, I think these two bullet points summarize the agreed desired outcome.

Makes sense. Would you mind updating the issue description to reflect this for easier reference?

semhustej, 2025-09-21:
> 
> Makes sense. Would you mind updating the issue description to reflect this for easier reference?

Done.