FreeCAD/FreeCAD #21865: Wiki is down - cannot access the database

semhustej, 2025-06-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

FreeCAD wiki is down. I got the same issue yesterday.

[image]

Tested from one PC and a mobile device with two different internet providers.

### Full version info

```shell
-
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

edragos, 2025-06-08:
I confirm.

Syres916, 2025-06-09:
Kurt has fixed this, see https://forum.freecad.org/viewtopic.php?t=97560 tested OK for me this morning

semhustej, 2025-06-09:
> Kurt has fixed this, see https://forum.freecad.org/viewtopic.php?t=97560 tested OK for me this morning

Yes, I can confirm it's fixed.

I am closing the issue.

closed 2025-06-09 by semhustej