FreeCAD/FreeCAD #21866: BIM: Arch_CurtainWall Swap Horizontal Vertical does not work

semhustej, 2025-06-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

`Arch_CurtainWall` `Swap Horizontal Vertical` parameter does not work. Steps to reproduce:

- open file (remove the .zip file extension)

[FC-cuetain-wall.FCStd.zip](https://github.com/user-attachments/files/20643880/FC-cuetain-wall.FCStd.zip)

- select any of the curtain walls (one is line based, the other is grid based)
- enable `Swap Horizontal Vertical` parameter
- nothing happens

See behavior in the video:

https://github.com/user-attachments/assets/6035bcb2-7d18-43e7-b8fb-0533fd5288d2

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Harvie, 2025-06-08:
Can confirm, i was planning to report this.

paullee0, 2025-06-08:
The property setting is never used in the code, needs to be fixed indeed :)

https://github.com/FreeCAD/FreeCAD/blob/593e9466392a9abb7c9960539761197a61719ce9/src/Mod/BIM/ArchCurtainWall.py#L170-L172

semhustej, 2025-06-08:
> The property setting is never used in the code, needs to be fixed indeed :)
> 

Thanks for investigating.

Frankly, I don't even think this property is needed, but that is outside of the scope of this issue report.

paullee0, 2025-06-15:
[image]
[image]

closed 2025-06-23 by yorikvanhavre

semhustej, 2025-06-27:
Tested with latest dev release. The issue is fixed, thank you.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.9477/1

```