FreeCAD/FreeCAD #21875: BIM: Arch_SectionPlane incorrect display style

semhustej, 2025-06-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

`Arch_SectionPlane` has incorrect display style. The problem is `Transparency` value for sure, I do not know whether other values too.

Steps to reproduce:

- open new file
- create `Arch_SectionPlane`
- the section plane is black. It has transparency set to 0.

[image]

I am not able to reproduce in FC 1.0.1. The section plane has transparency value 85 here.
I think this issue is not older than 30 days.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Anything else?

The issue was confirmed by another user in this forum thread: https://forum.freecad.org/viewtopic.php?p=830878

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-08:
@tetektoza might this be due to the RGBA/RGBT change in BIM?
or maybe @Roy-043 had some working plane related PRs?

maxwxyz, 2025-06-08:
@semhustej btw what is expected? The issue seems not to be in the handling of transparency as 0 = not transparent but rather in the changed default when creating a new plane?

Syres916, 2025-06-08:
It should look something like this (from 0.21.x):

[image]

ErtugrulEge, 2025-06-08:
From 1.0.1 version 

[image]

tetektoza, 2025-06-08:
@semhustej @maxwxyz Done. Was caused by this PR: https://github.com/FreeCAD/FreeCAD/pull/21342

Long explanation is that it was inheriting from a wrong object after the refactor, which had color predefined (black as you can see), which caused the properties to not be set properly (as we set them only if the object doesn't have a color already).

furgo16, 2025-06-09:
Apologies for the regression, and thank you @tetektoza for the analysis and the quick fix!

I did add tests to the refactoring PR to avoid this kind of situations. However, the starting point was nearly no tests, and the testing infrastructure needed to be fleshed out as well, thus many of the added tests were still fairly basic. I'm working on adding more sensible tests for each module. Starting with ArchWindow here: https://github.com/FreeCAD/FreeCAD/pull/21541

furgo16, 2025-06-09:
> please be careful also the default positions are different

@ErtugrulEge I'd perhaps wait until https://github.com/FreeCAD/FreeCAD/pull/21878 has been merged (or you can perhaps manually apply the fix) and if placement is still an issue after the fix, consider reporting it as a separate bug.