FreeCAD/FreeCAD #21882: CAM: Profile operation breaks with edges at same Z height as cut (revert 1b3205de20834482f8b347481a291bcb0626f821?)

dbtayl, 2025-06-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

https://github.com/FreeCAD/FreeCAD/commit/1b3205de20834482f8b347481a291bcb0626f821 added a check to ensure that open profiles are always cut at a Z height below the selected edges. This has the result of requiring the user to create extra geometry if the only edges for the desired profile are at the Z height to cut, which is not uncommon.

The commit says basically that it's a workaround for requiring the user to select a top edge for open profiles. Is that still accurate? It works fine with the simple test case below (reverted the above offending commit to get the profile to generate), but I don't know if there's still an issue there- one isn't linked in the commit.

Unless somebody can point to the issue this commit bandages, it actively hinders a normal workflow and should be reverted IMO.

Profile that would refuse to generate with the commit listed above, due to the selected edge being at the same Z height as the desired cut depth:
[image]

File (rename to .FCTstd)- if you regenerate the profile operation with the above commit, it will refuse to generate the path:
[tnut.zip](https://github.com/user-attachments/files/20647102/tnut.zip)

### Full version info

```shell
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.42096 (Git)
Build date: 2025/06/07 21:31:54
Build type: Unknown
Branch: main
Hash: 1b0ea183c203f23b67dc362a81f25f7aef78fea9
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1
Installed mods: 
  * OpenTheme 2024.9.1
  * FeedsAndSpeeds 0.6.0
  * btl 0.9.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-13:
@Russ4262 FYI

tarman3, 2025-06-13:
Probably this PR only add checks, but not change any logic
If remove this checks, select open profile and set depth below bottom profile or above top profile, you will get empty Path

dbtayl, 2025-06-14:
You're right- there's more going on here. See below for details, but:

- Open curves have inconsistent handling- they follow model edges when you change the final depth, but do _not_ follow model edges for multiple stepdowns. They break in various ways the above commit only half-bandages, notably when cutting above the model and when the model profile changes and the curve goes inside the model.
- Closed curves blindly project whatever curve you select to whatever depth you select, with zero regard for the model or stock. Closed curves will generate paths through the model, above/below the stock, whatever.

**So... what's the _desired_ behavior? I'll propose that "desired" in this case is making open curves be handled exactly like closed curves- a "dumb" projection, regardless of the model/stock.** Without actually looking at the code, I think that'd be a relatively quick fix that significantly improves consistency and usability.

I like a lot of things the open-edge profile is trying to do/doing, but IMO that's more a 3D operation, and unexpected behavior for "profile". It'd be great to morph that into a true waterline-style 3D toolpath, but that has additional complexity with handling ball-nose cutters.

Examples illustrating above points (with the above commit reverted):

This works fine:
[image]

Trying to cut lower generates no toolpath. Cutting higher works fine, up to the top of the model, where breaks:
[image]

That continues until I try to cut above the top of the stock, at which point it generates no toolpath again.

You get slightly different behavior with sloped faces. 3 different profile operations with the same edge selected at different depths... it follows the face?
[image]

More obviously following the face:
[image]

Note that it will NOT generate a profile at the same Z height as the edge- seems to break.

I don't even know if that's particularly undesired behavior, but it's inconsistent. If you have multiple stepdowns, it _doesn't_ follow the wall contour:
[image]

Worse, _closed_ curve behave completely differently, and blindly project whatever curve you select to whatever depth you select:
[image]

jffmichi, 2026-01-14:
.