FreeCAD/FreeCAD #21889: Sketcher: diameter constrain label position is wrong

pierreporte, 2025-06-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The problem started a few weeklies ago. Only The diameter constrain is affected (not radius, nothing else). You can clearly see that the box where the label should be is at the right place, but the text itself is pushed at the bottom, slightly out of it. There are even a few pixels at the very top of the box, like if the missing pixels are put there.

[image]

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/06/02 18:42:58
Build type: Release
Branch: main
Hash: 997d485124fab7b9ba971df59e91b55aeb5de61a
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/117.596/1
Installed mods: 
  * Curves 0.6.59
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.21
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-09:
I cannot reproduce. It looks a bit lower but not as bad as on your side:

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods: 
  * Cables 0.1.4
  * CfdOF 1.30.1
  * CurvedShapes 1.0.13
  * Curves 0.6.61
  * dodo 1.0.1
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Gridfinity 0.11.7
  * Manipulator 1.5.9
  * OpticsWorkbench 1.2.0
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.22
  * Silk 0.1.5
  * Telemetry 1.0.0

```

Syres916, 2025-06-09:
@pierreporte I have to say that looks like a packaging issue especially as you said it happened in the past few 'Weeklies'. Does the AppImage produced [here](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) give the same result, it uses slightly different libraries?

This is what I see on a self-compiled build:

[image]

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42134 (Git)
Build date: 2025/06/09 21:44:37
Build type: Release
Branch: main
Hash: ae3962191ac58a1fa63ed5f857e6f92c71766fb0
Python 3.13.2, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.1, boost 1_87, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9017/1
Installed mods: 
  * sheetmetal 0.7.22
  * Reinforcement
  * fasteners 0.5.38
```

PaddleStroke, 2025-08-25:
I cannot reproduce either.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.43047 +52 (Git)
Build date: 2025/08/25 08:08:50
Build type: Release
Branch: AstoCAD
Hash: b467872c42dd6ff7b02bcc0e84e72332aea448eb
Python 3.12.10, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, IfcOpenShell 0.8.1.post1, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/88.8003/1.25
```

Perhaps it's OS related?

PaddleStroke, 2025-08-26:
There is this issue (https://github.com/FreeCAD/FreeCAD/issues/22936)  that is similar, a text is being cropped and I can't reproduce that one either.

Maybe they are related

luzpaz, 2025-09-26:
@pierreporte this is still an issue correct ?

pierreporte, 2025-09-27:
Yes the problem is still there.

pierreporte, 2025-11-27:
I’ve tested with a fresh install on the same computer with the same result. I need to try the weekly again on my windows laptop but everytime I try it doesn’t start. I’ll try the RC.

maxwxyz, 2026-01-29:
@ipatch this issues needs to be bisected.

TomPcz, 2026-01-29:
This might be a coincidence, but PR #20540 was merged between the two versions. In commit 30fd8d6, the diameter symbol was changed from `QChar(216)` (Latin Capital Letter O with Stroke) to `QChar(0x2300)` (Diameter Sign).

I tried to build FreeCAD locally using pixi to test this. However, I cannot reproduce the issue with that build:
```
OS: openSUSE Tumbleweed (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.2.0dev.45184 (Git)
Build date: 2026/01/29 15:09:48
Build type: Release
Branch: main
Hash: b47bb6439875613fe966f539e9e03ba0b195f2ec
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/158.196/1
```

PaddleStroke, 2026-01-29:
I see, perhaps some font have some issue with QChar(0x2300) which cause excess padding or so.
Well maybe we change back to QChar(216). I checked in 1.0 and that character seems acceptable.

Or we find a fix. But  Ican't replicate so it's kind of hard to debug that.
Someone who can replicate and build can perhaps do some testing? Else we just merge reversing the character so people can feedback.

TomPcz, 2026-01-30:
I tried opening different fonts in KDE Font Viewer. Unless I am testing this incorrectly, it seems that many of them do not include the diameter symbol (shown on the left in the images), while O with Stroke (shown on the right) is almost always present. In that case, I have no idea where the diameter symbol displayed in Sketcher actually comes from.

Even if #26600 is merged, some form of font substitution will almost always take place for the diameter symbol. I will try to build that version and see whether I can replicate this issue by selecting different fonts.

**Helvetica** (downloaded from random website)
[image]

**Tahoma** (downloaded from random website)
[image]

**DejaVu Sans** (preinstalled on my system)
[image]

maxwxyz, 2026-01-30:
It is used in TechDraw too so maybe it can be implemented similar as in TD. Maybe use an SVG?

TomPcz, 2026-01-31:
I have been experimenting with this. I currently have _Helvetica_ installed (note that it is not installed by default on either Windows or any Linux distribution I use on my machines), so that is the font used for the labels. When _Helvetica_ was not installed, _Nimbus Sans_ was used instead.
```
> fc-match Helvetica 
Helvetica.ttf: "Helvetica" "Regular"
```
The diameter character, however, is not included in this font, so it appears to be substituted:
```
> fc-match Helvetica:charset=2300 
texgyreheros-regular.otf: "TeX Gyre Heros" "Regular"
```
Note that there is also the ◠ (U+25E0) Unicode character used for arc length. This one is not causing issues in my case, but I suspect it might if a problematic font is selected.
```
> fc-match Helvetica:charset=25E0 
FreeSans.otf: "FreeSans" "obyčejné"
```
So there are three different fonts involved, none of which are the hard-coded _Helvetica_ for most users.

To change which font is used when _Helvetica_ is not installed (on Linux), you can edit `~/.fonts.conf` to something like the following. _DejaVu Sans_ seems to include the diameter symbol, and when it is used, the label is aligned correctly. I assume a similar approach could be used to reproduce this issue: install _TeX Gyre Heros_ (which seems to be causing the problem, at least on my system) and set up the substitution to use that. Or try PR #26600, which I plan to build next.
```
<?xml version="1.0"?> 
<!DOCTYPE fontconfig SYSTEM "fonts.dtd"> 
<fontconfig> 
 <match target="pattern"> 
   <test qual="any" name="family"><string>Helvetica</string></test> 
   <edit name="family" mode="assign" binding="same"><string>DejaVu Sans</string></edit> 
 </match> 
</fontconfig>
```

pierreporte, 2026-02-01:
There was a discussion within the @FreeCAD/design-working-group to potentially provide fonts with FreeCAD for GUI instead of just osifont for TechDraw. Noto family was considered the most complete, as your list proves. Maybe it could be one solution.