FreeCAD/FreeCAD #21890: Sketcher: `Sketcher_ConstrainSymmetric` only works with points

pierreporte, 2025-06-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, it is only possible to make point symmetric around an axis or point. It should be possible to do that for all kind of shapes, especially lines.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/06/02 18:42:58
Build type: Release
Branch: main
Hash: 997d485124fab7b9ba971df59e91b55aeb5de61a
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/117.596/1
Installed mods: 
  * Curves 0.6.59
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.21
```

### Subproject(s) affected?

Sketcher

### Anything else?

It could be useful for `Sketcher_Symmetry` so it could constrain created geometry automatically.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-06-09:
What is needed:
  Geom/geom/line selection will auto-create point/point/line symmetry for each vertexes ?
or
  Create a new geom/geom/line symmetry which only show one constraint?

current implementation: user creates a point/point/line symmetry for each vertex
[image]

maxwxyz, 2025-06-09:
@pierreporte but couldn't the tool just apply a symmetry constraints to all points of the selected geometry? Or do you want only one visible constraint for the entire selected shape which can be deleted to break symmetry?

pierreporte, 2025-11-21:
@marbocub do you think you could tackle this with or after PR #25525?