FreeCAD/FreeCAD #21895: PartDesign: Inconsistent application of subtractive loft.

Maxwelldoug, 2025-06-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

With 2 sketches each containing 2 rectangles, I am attempting to subtractive loft of 2 bodies. With a direct copy paste of the 2 sketches from one to the other (they are defined relative to origin.)

When performing a subtractive loft, 'Body001' correctly cuts 2 channels, but 'Body' only cuts one, despite both cuts appearing in red in the tool before confirmation. Deleting the loft and recreating it while switching which sketch is the 'attachment' does not change the behaviour.

[A4-door.tar.gz](https://github.com/user-attachments/files/20657307/A4-door.tar.gz)

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.40267 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 04d2bcb)
Hash: 04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical/physical DPI: 96/188.749
Installed mods: 
  * OpenTheme 2025.5.20
  * boltsfc 2022.11.5
  * woodworking 0.21.2.33771.1.0.1
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Maxwelldoug, 2025-06-09:
[image]

FlachyJoe, 2025-06-09:
I can't confirm
```
OS: Debian GNU/Linux 13 (trixie) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41790 (Git)
Build date: 2025/05/19 19:40:38
Build type: Unknown
Branch: main
Hash: 2e4d7479a76c0bf85698b0aa10d47a1d393a5998
Python 3.13.3, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 8.0.0.rc1-7aa85582ad
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 85/86.4394/1
Installed mods: 
  * Render 2024.12.15
  * lattice2 1.0.0
  * Lithophane
  * Curves 0.6.62
  * fasteners 0.5.38
  * Assembly4 0.50.18
  * Telemetry 1.0.1
  * FreeCAD_SketchArch
  * Road 2025.2.22
  * FCPDWorkbench 0.4.0
  * sheetmetal 0.7.22
```

FEA-eng, 2026-10-16:
[This](https://github.com/FreeCAD/FreeCAD/issues/21895#issuecomment-4250427860) is still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```