FreeCAD/FreeCAD #21897: DXF: inconsistent object structure (Use layers: unchecked, Merge layers into blocks: checked)

furgo16, 2025-06-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

With these import setting, toggling between legacy (Python) and non-legacy (C++) importer,

[image]

Objects are imported differently:

[image]

Above is the C++ importer, which does not add groups, below is the Python importer, which effectively replaces the original layers with groups.

Ideally, the behaviour should be the same. If the "Use layers" checkbox is unchecked as in this example, I think the correct behaviour is that of the C++ importer.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42113 (Git) Snap 1486
Build date: 2025/06/08 20:11:05
Build type: Release
Branch: main
Hash: a38689b6fb94b1008f91508dd09f32610d7464aa
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.1083/1
Installed mods: 
  * OpenTheme 2025.5.20
  * Movie 2025.1.4
  * sheetmetal 0.7.22
  * fcVM.backup1749448662.6572056 (Disabled)
  * fasteners 0.5.38
  * ezydxf
  * Render 2024.12.15
  * Arch
  * Road 2025.2.22
  * Assembly4 0.50.18
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-06-10:
> Ideally, the behavior should be the same. If the "Use layers" checkbox is unchecked as in this example, I think the correct behavior is that of the C++ importer.
> 

I agree. But is it worth spending time fixing Lagacy Python importer? Especially with non critical (in my opinion) issue? I'd be interested in what the plan of the development team with regards to the legacy importer is.
I though the Python Importer will be deprecated once there is a parity with C++ importer (which is quite far away, but that is different issue).

> Also ideally, there should be one importer, but the reality is that often files fail to import with one, and it's good to have the other one as an alternative (I often have files failing to import with one, but I've rarely had an import fail on both).

This is already reported: https://github.com/FreeCAD/FreeCAD/issues/13599

semhustej, 2025-06-11:
> Which missing feature would you say is the most critical to get done?

I am not really sure. If I had to choose I would probably go for `Create Draft objects`.

By asking about plan I meant if work is still being done on Python Import/Export features.
I understand that there is no plan like 'C++ importer feature parity in 2025' :). It's clear that kpemartin has enough work for a long time just fixing bugs.

furgo16, 2025-06-13:
> By asking about plan I meant if work is still being done on Python Import/Export features.

Not features, just grave bug fixes, as far as I can tell.