FreeCAD/FreeCAD #21898: Cannot enter negative angle in Move / Array Transform

Beep6581, 2025-06-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The popup text box into which you type the desired angle does not accept "-" (negative) as the first character.

1. Sketch a shape.
2. Select it and press "W" to Move / Array Transform.
3. Two text boxes appear, one for offset and one for angle.
4. Type whatever as the offset, hit enter.
5. Focus moves to the angle text box. Try to type "-90" for the angle. It will ignore the "-" and only accept "90".
To work around this, after typing "90" move the caret to the beginning of the text box. Now when you type "-" it will accept.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42113 (Git)
Build date: 2025/06/08 20:11:05
Build type: Release
Branch: main
Hash: a38689b6fb94b1008f91508dd09f32610d7464aa
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5
Installed mods: 
  * OpenDark 2023.12.17
  * A2plus 0.4.68
  * fasteners 0.5.38
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.13
  * Curves 0.6.60
```

### Subproject(s) affected?

None

### Anything else?

[image]
(Note: -91.57 comes from my mouse movement, not from the keyboard)

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-06-11:
Cannot replicate on 

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42081 +50 (Git)
Build date: 2025/06/10 09:11:25
Build type: Release
Hash: 34bbb7d847064cee87a215e401c5385dda9de6a1
Python 3.12.10, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, IfcOpenShell 0.8.1.post1, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/88.8003/1.25

There's a recent PR on OVP in latest weekly so perhaps that's why. I will rebase and test again.

markit, 2025-06-17:
I can confirm that in 1.1dev with Sketcher i.e. Offset command does not accept negative values from keyboard too, seems that they are rejected (some sort of "input filtering"). Also I've troubles with, in this case positive, values but with decimal separator (in Italy is ",", I've set locales and preferences to use it). If I want to set an offset i.e. '0,5' as soon as I type "0" I've the error message tip that the offset value can't be "0". If I try "1,5" as soon as I enter "1," the field becomes completely selected so if I then type "5" the field becomes "5" (seems that it ignores the locale and considers "," as a wrong character".
This is a regression since both (negative values and decimal separator) it worked fine with 1.0.
Since I'm only recently playing with Sketcher, don't know when those problems were introduced. Sure I have confirmed with FreeCAD_weekly-builds-42095-conda-Linux-x86_64-py311.AppImage and onwards.

WORK FINE with:
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

NOT WORKING WITH, I.E.:
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.42180 (Git) AppImage
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/161.962/1

closed 2025-06-20 by hyarion

Tammy-jess, 2025-06-24:
This still doesn't seem to be rolled out to the latest x86_64 appimage.. which remains as build-42180 for windows & linux, while there is build-42203 available for linux aarch64 and both MacOS silicons (x86 and arm64).

For my knowledge — are these indeed built weekly (automagically, or manually?), and if so, what day do the new builds for x86_64 drop? 

Cheers, Tammy

Tammy-jess, 2025-06-26:
Can confirm here also.  -.5 works, .5 works .. but entry box does not accept zero as a singular character  (Hey, we're saving keystrokes!)   

Probably not intended behaviour though  :)

We -can- enter the zero, after the entry contains a valid number by back-arrowing to the start.