FreeCAD/FreeCAD #21903: QObject::disconnect: wildcard call disconnects from destroyed signal of QObject

Harvie, 2025-06-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

After upgrade to qt-6.9.1, FreeCAD GUI is intermittently unresponsive and prints lot of these warnings all the time:

```
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
QObject::disconnect: wildcard call disconnects from destroyed signal of QObject::134471623393408
```

Allegedly it was ok with qt-6.9.0
https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/issues/21

Other Qt applications seem to be unaffected.

Please see qt-6.9.1 changelog to make sure they have not broken anything FreeCAD related:
https://code.qt.io/cgit/qt/qtreleasenotes.git/about/qt/6.9.1/release-note.md

### Full version info

```shell
OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-10:
@kadet1090

Syres916, 2025-06-11:
Confirmed, on initialisation the first of these Qt warnings is triggered in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Tux/PersistentToolbarsGui.py#L29 which is a very common line of code.

closed 2025-06-23 by hyarion