FreeCAD/FreeCAD #21904: Double-clicking on boolean property checkboxes blocks all properties

FEA-eng, 2025-06-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

#21555 implemented checkboxes instead of true/false drop-down list for boolean properties all over FreeCAD. I've noticed that double-clicking on them blocks all properties in the task panel.

Steps to reproduce:
1. Add Part Cube
2. Double-click on the Bounding Box property.
3. Now you can't change any property in the task panel.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

m1cha1, 2025-06-10:
Confirmed on

> OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1

After you have blocked the checkbox, you can click on the model to unlock the checkboxes.

Syres916, 2025-06-10:
I've not run a Bisect but https://github.com/FreeCAD/FreeCAD/commit/272c3034b44c6e36df72ff56d76ae75ff9454372 might need a look at.
@3x380V do we need something like ignoring double clicks in the PropertyEditor or at least only where appropriate but I can't think of one?

3x380V, 2025-06-10:
As a quick test I returned to 7f6af23acb4e5e66c43dc41a6a803d91f87e6233, the last one before #21659. Double click still behaves strange way, but a bit different - it still blocks, but sometimes it can be unblocked with the "right" clicking timing. I'll investigate more later.

Syres916, 2025-06-11:
@FEA-eng if you can compile the PR on Linux and see if you can spot any regressions, I used Fasteners as there's checkboxes, combobox & spinboxes in most of the objects.

closed 2025-06-13 by kadet1090