FreeCAD/FreeCAD #21909: PartDesign Pocket with multiple solids produces a good result with a bad operation.

leoheck, 2025-06-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Weird thing happened.
I have a part cylinder with Radius (let's say 30)
I want to cut it in half by sketching a bigger rectangle than the cross-section of the cylinder. 
Then I use PartDesign Pocket.
When I go to that feature, which starts with a fixed length of 5 mm.
This initial result produces 2 solids.
The length is wrong, but since it produces 2 solids, one is removed.
Then I have a bad feature set that produces the expected result.
And this is a bit weird, isn't it?

When this happens, a message is displayed, but not everyone will see this since the operation is good.
```
04:12:23  Transformed: Result has multiple solids. Only keeping the first.
```

Is this an expected behavior? I feel this is going to cheat users, and they will end up having poorly designed parts.

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-06-11:
> The length is wrong, but since it produces 2 solids, one is removed. Then I have a bad feature set that produces the expected result. 

I am assuming the experimental "Allow Compound" was set to False on the body. Otherwise it would show both remaining parts of the original cylinder?

FEA-eng, 2026-01-14:
@leoheck Please confirm if the issue is still relevant.

closed 2026-01-14 by maxwxyz