FreeCAD/FreeCAD #21918: BIM:  Part ExplodeCompound tool is missing from General modification tools bar

markkomartin, 2025-06-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

BIM workbench has make compound tool in General modification tools bar. 
An inexperienced user might be confused in not finding Part ExplodeCompound tool.
You can make a custom toolbar from (Tools->Customize) and select explode compound from Part WB but it would be better to have it as default.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

### Subproject(s) affected?

None

### Anything else?

[image]

forum thread  https://forum.freecad.org/viewtopic.php?p=831208#p831208

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

markkomartin, 2025-06-12:
Wow good catch @Roy-043 .
There is actually no need for a Part ExplodeCompound tool when you can simply delete the compound.

At this point I would just add a note to the Part Compound wiki page.
I am not subscribed to the site https://wiki.freecad.org/ yet. 
If you agree I will try to subscribe and add the note. 

Thanks again for the time dedicated to this issue.

luxurystorecc1, 2025-07-04:
https://airjordan-1.com/ Air Jordan 1
https://wholesalejerseysfreeshipping.com/ Cheap NFL Jerseys
https://www.nfljerseyss.us.com/ NFL Jerseys
https://airjordan-4.com/ Air Jordan 4
http://peter-milla.com/ Peter Milla
http://yeezyboost-350.com/ Yeezy 350
https://pandoracharmsbracelet.com/ Pandora Charms
https://adidasyeezyslides.com/ Yeezy Slides
https://pandora-bracelets.com/ Pandora Bracelets

Pandora Bracelets  | Official Handcrafted Jewelry
Official Pandora Bracelets – Free Delivery Available