FreeCAD/FreeCAD #21922: CAM: Start coolant only after the tool back to the workpiece

tarman3, 2025-06-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

> After changing the tool, the spindle moves to the tool length sensor and measures the tool length offset. The spindle is then started immediately and the coolant is also started. It would be better if at least the coolant did not start until the tool was back over the workpiece.

> Can the postprocessor be adapted to this? I have always adapted the .ngc file manually with an editor and moved the M7/M8 command.

This text from from [forum](https://forum.freecad.org/viewtopic.php?t=97472)

@LarryWoestman
I done some fix for old LinuxCNC postprocessor, but probably this changes should be incorporated into the refactored post-processors

@sliptonic 
should I create pull request for old LinuxCNC postprocessor?

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: SearchBar 1.4.2 (Disabled); CurvedShapes 1.0.13; lattice2 1.0.0; Silk 0.1.9; freecad.gears 1.3.0; MeshRemodel 1.10.37; Curves 0.6.63; sheetmetal 0.7.22; SaveAndRestore 0.2.1 (Disabled); Defeaturing 1.2.6
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

LarryWoestman, 2025-06-11:
This is a bit tricky, I think.  Sometimes there is a multi-second delay from the time you start the coolant until the coolant is actually flowing at a reasonable rate.  Sometimes the coolant starts to flow almost immediately.  So one choice doesn't fit all controllers or pumps.

The "safest" assumption is to start the coolant as soon as you can tell that you might need it, which is when the spindle starts turning.  You might get more coolant sooner than you need it, but you try to avoid running your cutter "dry" for any longer than you can help.

Ideally there would be some concept of "coolant delay" where the coolant is started "early enough" so that coolant is flowing by the time the spindle is up to speed to prevent "dry" cutting.  But that is way too complicated for FreeCAD.  That would be difficult to figure out and provide a GUI for, since it would "overlap" with several other operations.

Another option is not to start the spindle (and the coolant) until the tool is back over the workpiece.  But now we have the same problem with having to wait for the spindle to come up to full speed before the cutter starts cutting.  Some of the postprocessors have a "delay after spindle start" option that puts in a specified "pause" after the spindle start g-code instruction and before any other instructions are processed.

So I suggest that rather than trying to guess when to start the coolant we use the delay after the spindle start command to make sure that the spindle is up to speed and that the coolant is flowing before moving on to cutting.  The original problem is that the spindle is started while it is over the tool length sensor and not when it is back over the work.

LarryWoestman, 2025-06-30:
I agree with what @Connor9220 has said, with additional emphasis on being safe.  There are times when the human might know more about the situation than FreeCAD can assume, but trying to provide options that sometimes work and sometimes aren't safe in FreeCAD is risky.

It is sometimes reasonable to hand-edit the generated g-code to take advantage of the additional information that the human might have.  That isn't very practical if there is a lot of g-code to edit.