FreeCAD/FreeCAD #21927: Sketcher: OVP regression: pressing enter validate the tool

PaddleStroke, 2025-06-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

- start rectangle tool.
- type 50 for the width
- press enter to validate it.

Expected: Goes to the height spinbox
What happens now: validate the rectangle creation.

@kadet1090 

https://github.com/user-attachments/assets/f1e28200-7fc6-4c8a-a3e4-c792459bc9fe

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42139 +51 (Git)
Build date: 2025/06/11 07:26:57
Build type: Release
Branch: AstoCAD
Hash: 3d656f9fb970d0658f24da9811b9be2c6a519ae5
Python 3.12.10, Qt 6.8.3, Coin 4.0.3, Vtk 9.4.2, boost 1_87, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.5, IfcOpenShell 0.8.1.post1, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/88.8003/1.25
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kadet1090, 2025-06-11:
As far as I remember there was a long discussion about intended behavior, I don't remember right now outcomes of it.

@FreeCAD/design-working-group @tetektoza

pierreporte, 2025-06-11:
I think that the proposed behavior makes sense.

Roy-043, 2025-06-12:
I notice an issue that may be related:
1. Set OVP preference to "Position and dimensions".
2. Create a new sketch.
3. Start Sketcher_CreateLine.
4. Enter the X coordinate of the start point.
5. Press Tab to switch to the Y input box.
6. Start entering a numerical value with multiple digits.
7. Result: after pressing the key for the first digit the input gets confirmed.

```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

tetektoza, 2025-06-12:
@Roy-043 ah right, thanks. I can confirm the issue and it's a regression, but not related to this topic. I will fix it separately.

closed 2025-06-23 by hyarion